stack-of-tasks/tsid #7: definitions not included in header file

proyan, 2017-07-12:
The definitions of tsid::solvers::SolverHQuadProgRT are in the hxx file, which are not included when one includes the hpp header. So at runtime, since the symbols are not defined, there is an undefined symbol error.

I tried to include <tsid/solvers/solver-HQP-eiquadprog-rt.hxx> in addition to the hpp file, and it gives me the following error: 
`<INSTALL>/include/tsid/solvers/solver-HQP-eiquadprog-rt.hxx:149:11: error: ‘set_is_malloc_allowed’ is not a member of ‘Eigen::internal’
           EIGEN_MALLOC_ALLOWED
           ^
<INSTALL>/include/tsid/solvers/solver-HQP-eiquadprog-rt.hxx:151:11: error: ‘set_is_malloc_allowed’ is not a member of ‘Eigen::internal’
           EIGEN_MALLOC_NOT_ALLOWED
           ^
<INSTALL>/include/tsid/solvers/solver-HQP-eiquadprog-rt.hxx:175:7: error: ‘set_is_malloc_allowed’ is not a member of ‘Eigen::internal’
       EIGEN_MALLOC_ALLOWED`

And I can't find the macros EIGEN_MALLOC(_NOT)_ALLOWED in the eigen header directory
Any clues on how to solve this?

jcarpent, 2017-07-12:
The idea of .hxx is to replace the previous .cpp file. So if you just need the declaration of the entity, please use .hpp. If you need also the definition of the methods, please use .hxx.

Anyway, can you give me an example of non-compiling file?

jcarpent, 2017-07-12:
The macros **EIGEN_MALLOC(_NOT)_ALLOWED** are defined in `math/fwd.hpp`

proyan, 2017-07-12:
set_is_malloc_allowed is defined in the eigen macro EIGEN_RUNTIME_NO_MALLOC. This macro should be set before using this function.

jcarpent, 2017-07-12:
Maybe it certainly comes from your version of Eigen. Can you check? This is not the case for all the Eigen versions.

proyan, 2017-07-12:
I am on Eigen 3.2.0 (the version provided by ubuntu14.04)

jcarpent, 2017-07-12:
Ok. I'm going to fix the bug related to older Eigen version.

jcarpent, 2017-10-09:
solved by #8.

closed 2017-10-09 by jcarpent