stack-of-tasks/tsid #12: devel stalled

olivier-stasse, 2018-01-30:
Dear all,
The devel branch is stalled and behind master.
I suggest to remove the devel branch if this is fine with everyone.

jcarpent, 2018-01-30:
I would suggest to keep the devel branch for developments and master for stable versions.
In other words, we can align both branches and make a minor release of the current version and push it on robotpkg.

olivier-stasse, 2018-01-30:
Exactly and for this reason the _devel_ branch should be ahead of master not the other way around. For it means that _devel_ is not used for development and is currently meaningless.

olivier-stasse, 2018-01-30:
Or you mean that devel was used to develop a specific topic. Thus I would recommend to rename it as a specific topic for book keeping and restart a new devel branch from master.

jcarpent, 2018-01-30:
I just meant that devel should be used for development and keep master a bit static, similarly as we do for pinocchio or eigenpy for instance.

olivier-stasse, 2018-01-30:
Dear Justin, Thanks for the clarification.

closed 2018-01-30 by olivier-stasse