stack-of-tasks/tsid #23: Not compilable with the new version of Pinocchio

huaijiangzhu, 2019-01-24:
Hi,

I am getting errors such as: 
error: ‘se3’ does not name a type
It looks like tsid is not adapted to the new namespace in Pinocchio, would you be able to fix this?

huaijiangzhu, 2019-01-24:
> @huaijiangzhu You just need to add this compilation flag `-DPINOCCHIO_ENABLE_COMPATIBILITY_WITH_VERSION_1`when compiling Pinocchio, nothing more.
> 
> Then someone has to take care of making TSID compatible with the last version of Pinocchio. @huaijiangzhu Can you do it? It is not hard, a simple **find and replace**.

Thanks for the reply! I just replaced the namespace but now I got a new error:
error: no matching function for call to ‘centerOfMass(const Model&, tsid::robots::RobotWrapper::Data&, int, bool)’
       pinocchio::centerOfMass<true,true,true>(m_model, data, false);

Looks like the centerOfMass API is also changed in the new Pinocchio, but I'm not sure how to modify it accordingly. Can you help me with that?

jcarpent, 2019-01-24:
Yes, sure. For this function, the template parameters have been replaced by a `LEVEL`, see https://github.com/stack-of-tasks/pinocchio/blob/01f211eceda3ac2e5edc8cf101690afb6f3184d3/src/algorithm/center-of-mass.hpp#L101 for further details.
In your case, you want to compute the center of mass position, velocity, and acceleration together, so then LEVEL = 2.

jcarpent, 2019-01-24:
Because you have to remove the template parameters `<true,true,true>` to obtain:
```
pinocchio::centerOfMass(m_model, data, 2, false);
```

huaijiangzhu, 2019-01-24:
@jcarpent Thanks! Now the code compiles. But when I try to run the test, I get an import error:

Traceback (most recent call last):
  File "test_Formulation.py", line 2, in <module>
    import tsid
  File "/home/hz/py27/local/lib/python2.7/site-packages/tsid/__init__.py", line 1, in <module>
    import libtsid_pywrap as tsid
ImportError: libtsid.so: cannot open shared object file: No such file or directory
Any idea what can cause this?

jcarpent, 2019-01-25:
Have you installed TSID? In which directory?
Similarly to Pinocchio, you have to configure your environment variables. You can look at https://stack-of-tasks.github.io/pinocchio/download.html#Install_4

jcarpent, 2019-01-25:
@huaijiangzhu Can you make a PR to fix these issues that you've raised?

huaijiangzhu, 2019-01-25:
> Similarly to Pinocchio, you have to configure your environment variables. You can look at https://stack-of-tasks.github.io/pinocchio/download.html#Install_4

Thanks, it works now!

> @huaijiangzhu Can you make a PR to fix these issues that you've raised?

Sure, I will!