stack-of-tasks/tsid #24: Installation problem with pinocchio dependency

ascane, 2019-01-25:
I followed Pinocchio's installation procedure and installed it on Ubuntu 16.04. Then, I followed the tsid's installation guide. However, `ADD_REQUIRED_DEPENDENCY("pinocchio >= 1.3.0")` in `CMakeLists.txt` does not seem to be satisfied.

After  `cmake .. -DCMAKE_BUILD_TYPE=RELEASE -DCMAKE_INSTALL_PREFIX=$DEVEL/openrobots`, I got the following error message.

```
-- Option EIGEN_RUNTIME_NO_MALLOC on.
-- eigen3 >= 3.2.0 is required.
-- Checking for module 'eigen3 >= 3.2.0'
--   Found eigen3 , version 3.2.92
-- Pkg-config module eigen3 v3.2.92 has been detected with success.
-- pinocchio >= 1.3.0 is required.
-- Checking for module 'pinocchio >= 1.3.0'
--   
CMake Error at /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:367 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:532 (_pkg_check_modules_internal)
  cmake/pkg-config.cmake:288 (PKG_CHECK_MODULES)
  cmake/pkg-config.cmake:459 (ADD_DEPENDENCY)
  CMakeLists.txt:76 (ADD_REQUIRED_DEPENDENCY)

-- Configuring incomplete, errors occurred!
See also "/opt/openrobots/src/tsid/_build-RELEASE/CMakeFiles/CMakeOutput.log".
```

`apt list --installed | grep pinocchio`
gave me
```
robotpkg-pinocchio/xenial,now 1.3.3 amd64 [installed,automatic]
robotpkg-py27-pinocchio/xenial,now 1.3.3 amd64 [installed]
```

Any help would be appreciated.

jcarpent, 2019-01-25:
I guess you have not configured your environment variables.
You can find the instructions here: https://stack-of-tasks.github.io/pinocchio/download.html

ascane, 2019-01-25:
Thanks for your help. I did configure my environment variables PATH, PKG_CONFIG_PATH, LD_LIBRARY_PATH, PYTHONPATH (and also checked with `echo $PKG_CONFIG_PATH`, etc). Are there any other environment variables that need to be changed?

andreadelprete, 2019-01-25:
Try executing `pkg-config --cflags pinocchio` to see if it finds it

ascane, 2019-01-25:
`-DURDFDOM_TYPEDEF_SHARED_PTR -DWITH_URDFDOM -DWITH_HPP_FCL -DFCL_USE_ASSIMP_UNIFIED_HEADER_NAMES -DFCL_HAVE_OCTOMAP= -I/opt/openrobots/include -I/usr/include/assimp -I/usr/include/eigen3`

jcarpent, 2019-01-25:
Yes, indeed. Normally the rest should be fine.