stack-of-tasks/tsid #28: A virtual box for TSID, pinocchio v2 branch

nmansard, 2019-01-28:
I created a VirtualBox image for trying Pinocchio V2. It is at this address:
https://github.com/nmansard/wsmemmo_pinocchio

You can modify this VirtualBox image to make TSID run on it. Here are the steps for making that happens.
git clone --recursive https://github.com/stack-of-tasks/tsid
cd tsid
mkdir build
cd build
export PKG_CONFIG_PATH=$PINOCCHIO_PATH/lib/pkgconfig:$PKG_CONFIG_PATH
cmake .. -DCMAKE_INSTALL_PREFIX=$PINOCCHIO_PATH
make -sj2 install

andreadelprete, 2019-01-28:
Updated version of the script:
```
sudo apt-get remove robotpkg-pinocchio
sudo apt-get install python-tk
git clone --recursive https://github.com/stack-of-tasks/tsid
cd tsid
git checkout pinocchio_v2
mkdir build
cd build
export PKG_CONFIG_PATH=$PINOCCHIO_PATH/lib/pkgconfig:$PKG_CONFIG_PATH
cmake .. -DCMAKE_INSTALL_PREFIX=$PINOCCHIO_PATH -DCMAKE_BUILD_TYPE=RELEASE
make -sj2 install
```

andreadelprete, 2019-01-28:
Beware that the first two `apt-get` commands do not work well if you put them in a bash script, so please run them one by one by copy-pasting them in the terminal.

After you have successfully compiled and install TSID you can test it by executing:
```
cd tsid/exercizes
python ex_1.py
```
If it does not work the first time, try to execute it a second time (sometimes the viewer crashes for mysterious reasons)

closed 2019-01-30 by andreadelprete