stack-of-tasks/tsid #31: Release & Pinocchio v2

nim65s, 2019-02-12:
Hi @andreadelprete 

Following discussions in https://github.com/stack-of-tasks/pinocchio/pull/645 I would like to know what is the current status of this package. 
Is it ready for a new release, or are you still working on it ?
I can make the release if you want, adressing #20 in the same time.

And is the pinocchio_v2 branch ready ? If so I could also merge it for the release, and publish it to robotpkg at the same moment as every other of our projects that depend on hpp-fcl / pinocchio.

andreadelprete, 2019-02-12:
Hi @nim65s , after a quick look at the (really) long discussion you had with the others, I wonder whether this really concerns me and TSID. TSID depends on pinocchio, but not on FCL because it doesn't use the collision checking features. So, should I be concerned with this discussion?

To answer your question I think TSID is ready for a new release, using the branch pinocchio_v2, which you can merge into master.

andreadelprete, 2019-02-12:
Regarding the BSD license issue #20, TSID contains the software [eiquadprog](https://github.com/stack-of-tasks/tsid/blob/master/include/tsid/solvers/eiquadprog_2011.hpp#L69) which is under GPL. This might be an issue, doesn't it?

nim65s, 2019-02-12:
Thanks for your answers. If the branch pinocchio_v2 is ready, this is all we need.

olivier-stasse, 2019-02-12:
@andreadelprete When the planning messed up (or when there is no planning) avoiding collision in the control is the nice feature...
Yes having eiquadprog inside TSID is an issue.

jcarpent, 2019-02-12:
@olivier-stasse I will provide a solution for that as soon as possible to remove the strong dependency on EiquadProg.

andreadelprete, 2019-02-12:
> @andreadelprete When the planning messed up (or when there is no planning) avoiding collision in the control is the nice feature...

sure, eventually it would be nice to add collision avoidance in TSID. I only meant that currently it's not there, so TSID wouldn't be affected by any change in HPP-FCL.

olivier-stasse, 2019-03-02:
Regarding the compatibility with pinocchio-v2 I worked on a PR to update the library to pinocchio-v2.1.0.
I made a mistake and pushed directly on devel instead of my PR. Let me know if this is  a problem, I will revert it if needed.

andreadelprete, 2019-03-02:
Hi @olivier-stasse, as far as I know TSID is already compatible with Pinocchio-v2. At the very least the branch `pinocchio-v2` is. Moreover, @nim65s should have merged that branch into master a few weeks ago, so even master should be aligned with pinocchio-v2 now.

andreadelprete, 2019-03-02:
Actually after a quick look at the code it seems that it is `devel` to be aligned with pinocchio-v2. Master is still on pinocchio-v1.

olivier-stasse, 2019-03-02:
Hi @andreadelprete. I did some tests today and some stuff were not working. For instance there was a lot of se3 namespace instead of pinocchio.

olivier-stasse, 2019-03-02:
And i am talking about the devel branch. But maybe I missed something

nim65s, 2019-03-03:
https://github.com/stack-of-tasks/tsid/pull/25 changed the namespace, but in master instead of devel.

I will handle that tomorrow, and make a release of TSID.

olivier-stasse, 2019-03-03:
Ok, sorry my bad. Looking at #25, it seems that the branch pinocchio_v2 doing the right fix was not merged into devel.

olivier-stasse, 2019-03-03:
@nim65s for the next release, we could:
- integrate the PRs that were not included
- fix the warnings
I'll try to make a proposal in the devel branch and compare with the branch pinocchio_v2

nim65s, 2019-03-04:
@olivier-stasse So do you want to handle the merge in #33 ?

nim65s, 2019-03-04:
Or should we just forget about either the pinocchio_v2 branch, or @olivier-stasse 's last commits to devel ?

nim65s, 2019-03-04:
Well… For me, pinocchio_v2 branch is building (and tests are passing) and not devel, so I think I will use this one for the release.

nim65s, 2019-03-04:
Everything now looks good to me. @andreadelprete @olivier-stasse can you just review this commit: https://github.com/stack-of-tasks/tsid/pull/33/commits/dab03d38dabd6c51677bace5373c7fb936ee7e8d?w=1 ?

nim65s, 2019-03-04:
Github overrides the link… You will have to go there https://github.com/stack-of-tasks/tsid/pull/33 and then click on "Merge branch 'devel' into pinocchio_v2"

andreadelprete, 2019-03-04:
> Everything now looks good to me. @andreadelprete @olivier-stasse can you just review this commit: [dab03d3](https://github.com/stack-of-tasks/tsid/commit/dab03d38dabd6c51677bace5373c7fb936ee7e8d) ?

it looks good to me!

olivier-stasse, 2019-03-05:
Good to me too

closed 2019-03-06 by nim65s