stack-of-tasks/tsid #37: Problem with constructor delegration for Contact6D on macOS.

jviereck, 2019-04-08:
Compiling current master (813bea98aab69b0) on macOS is broken. This seems to the stricter error / spec handling of clang over gcc. The error is:

```
o CMakeFiles/tsid.dir/solvers/solver-HQP-factory.cpp.o -c /Users/jviereck/dev/tools/tsid/src/solvers/solver-HQP-factory.cpp
/Users/jviereck/dev/tools/tsid/src/contacts/contact-6d.cpp:64:5: error: delegating constructors are permitted only in C++11
    Contact6d(name, robot, frameName, contactPoints, contactNormal, frictionCoefficient, minNormalForce, maxNormalForce)
    ^~~~~~~~~
```

When enabling C++11 in `CMakeLists.txt` as

```
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")
```

things compile properly.

jviereck, 2019-04-08:
> When enabling C++11 in CMakeLists.txt as
>
> set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")
>
> things compile properly.

This makes the source compile. However, when constructing the tsid.RobotWrapper from python, this will give:

```
robot = tsid.RobotWrapper(urdf, vector, se3.JointModelFreeFlyer(), False)

ArgumentError                             Traceback (most recent call last)
<ipython-input-25-1865ae3203f2> in <module>()
----> 1 tsid.RobotWrapper(urdf, vector, se3.JointModelFreeFlyer(), False)

ArgumentError: Python argument types in
    RobotWrapper.__init__(RobotWrapper, str, StdVec_StdString, JointModelFreeFlyer, bool)
did not match C++ signature:
    __init__(_object*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > filename, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > package_dir, boost::variant<pinocchio::JointModelRevoluteTpl<double, 0, 0>, pinocchio::JointModelRevoluteTpl<double, 0, 1>, pinocchio::JointModelRevoluteTpl<double, 0, 2>, pinocchio::JointModelFreeFlyerTpl<double, 0>, pinocchio::JointModelPlanarTpl<double, 0>, pinocchio::JointModelRevoluteUnalignedTpl<double, 0>, pinocchio::JointModelSphericalTpl<double, 0>, pinocchio::JointModelSphericalZYXTpl<double, 0>, pinocchio::JointModelPrismaticTpl<double, 0, 0>, pinocchio::JointModelPrismaticTpl<double, 0, 1>, pinocchio::JointModelPrismaticTpl<double, 0, 2>, pinocchio::JointModelPrismaticUnalignedTpl<double, 0>, pinocchio::JointModelTranslationTpl<double, 0>, pinocchio::JointModelRevoluteUnboundedTpl<double, 0, 0>, pinocchio::JointModelRevoluteUnboundedTpl<double, 0, 1>, pinocchio::JointModelRevoluteUnboundedTpl<double, 0, 2>, boost::recursive_wrapper<pinocchio::JointModelCompositeTpl<double, 0, pinocchio::JointCollectionDefaultTpl> > > roottype, bool verbose)
    __init__(_object*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > filename, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > package_dir, bool verbose)
```

It look the type `se3.JointModelFreeFlyer()` is not detected properly during the construction. Maybe this happens because pinocchio is not compiled with C++11, but I am not sure.

andreadelprete, 2019-04-09:
Yes, I think I remember seeing this problem. I guess that with gcc I've managed to find a workaround, probably by enabling the C++11 flag, which on Ubuntu didn't lead to the runtime error you had. Maybe the easiest way out is to avoid the `delegating constructors` in contact-6d so that you can compile tsid without the c++11 flag. Could you give that a try? You could simply add a private initialization method in contact6d which is called by both its constructors, so that we don't have to duplicate code.

jviereck, 2019-04-10:
Fixed by #38.

PR #38 (2019-04-09): Avoid delegation costructor in Contact6d as it is a C++11 feature.
Fixes #37.
commit e7cceed3 (2019-04-09): Avoid delegation costructor in Contact6d as it is a C++11 feature. Fixes #37
closed 2019-04-10 by jviereck