stack-of-tasks/tsid #45: Definition of Contact Points

proyan, 2019-04-18:
Hi all,

I started to look at the Jupyter notebook (ex1), to understand how the python API works. I was looking at the definition of contact points, and I realize that the value of lz is taken as a positive number (0.107) (https://github.com/stack-of-tasks/tsid/blob/devel/exercizes/notebooks/TSID_ex1.ipynb). If the contact points are defined on the sole, lz should be negative.
Am I missing something?

andreadelprete, 2019-04-18:
You may be right but I don't remember by heart. You can check how they are defined in hrp2-torque-control, or you could visualize them in the viewer to check visually.

proyan, 2019-04-19:
Thanks Andrea for the reply. I'll try to find out.
I'll close this issue.

closed 2019-04-19 by proyan