stack-of-tasks/tsid #50: signature C++ problem with python bindings

olivier-stasse, 2019-10-11:
With robotpkg install and pinocchio 2.1.7 when trying to do
```
cd exercizes
python ex_1.py
```
I have the following error:
```
Traceback (most recent call last):
  File "ex_1.py", line 15, in <module>
    tsid = TsidBiped(conf)
  File "/home/ostasse/devel-src/SoT/tsid/exercizes/tsid_biped.py", line 24, in __init__
    self.robot = tsid.RobotWrapper(conf.urdf, vector, pin.JointModelFreeFlyer(), False)
Boost.Python.ArgumentError: Python argument types in
    RobotWrapper.__init__(RobotWrapper, str, StdVec_StdString, JointModelFreeFlyer, bool)
did not match C++ signature:
    __init__(_object*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > filename, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > package_dir, boost::variant<pinocchio::JointModelRevoluteTpl<double, 0, 0>, pinocchio::JointModelRevoluteTpl<double, 0, 1>, pinocchio::JointModelRevoluteTpl<double, 0, 2>, pinocchio::JointModelMimic<pinocchio::JointModelRevoluteTpl<double, 0, 0> >, pinocchio::JointModelMimic<pinocchio::JointModelRevoluteTpl<double, 0, 1> >, pinocchio::JointModelMimic<pinocchio::JointModelRevoluteTpl<double, 0, 2> >, pinocchio::JointModelFreeFlyerTpl<double, 0>, pinocchio::JointModelPlanarTpl<double, 0>, pinocchio::JointModelRevoluteUnalignedTpl<double, 0>, pinocchio::JointModelSphericalTpl<double, 0>, pinocchio::JointModelSphericalZYXTpl<double, 0>, pinocchio::JointModelPrismaticTpl<double, 0, 0>, pinocchio::JointModelPrismaticTpl<double, 0, 1>, pinocchio::JointModelPrismaticTpl<double, 0, 2>, pinocchio::JointModelPrismaticUnalignedTpl<double, 0>, pinocchio::JointModelTranslationTpl<double, 0>, pinocchio::JointModelRevoluteUnboundedTpl<double, 0, 0>, pinocchio::JointModelRevoluteUnboundedTpl<double, 0, 1>, pinocchio::JointModelRevoluteUnboundedTpl<double, 0, 2>, pinocchio::JointModelRevoluteUnboundedUnalignedTpl<double, 0>, boost::recursive_wrapper<pinocchio::JointModelCompositeTpl<double, 0, pinocchio::JointCollectionDefaultTpl> >, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_, boost::detail::variant::void_> roottype, bool verbose)
    __init__(_object*, pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> Pinocchio Model, bool verbose)
    __init__(_object*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > filename, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > package_dir, bool verbose)
```
I identified the problem as being a new wrapping from pinocchio of the JointModelVariant.
I will try to propose a PR asap.

olivier-stasse, 2019-10-11:
TSID python bindings of the robotwrapper needs something like:
https://github.com/stack-of-tasks/pinocchio/blob/380fdefb172efa07f2dd9112ce2d389e25aa7a41/bindings/python/parsers/parsers.hpp#L66

andreadelprete, 2019-10-13:
This is solved by the PR of @olivier-stasse so I close the issue

closed 2019-10-13 by andreadelprete