stack-of-tasks/tsid #52: compiling problem

xuhuairuogu, 2019-10-12:
Problem Description
   My OS is Ubuntu18.04.
   First, I installed PInocchio successfully following the instructions on this website https://stack-of-tasks.github.io/pinocchio/download.html :
  1. Add robotpkg apt repository
  2. Install Pinocchio
    sudo apt install robotpkg-py27-pinocchio
  3. Configure environment variables(add the following to the file ~/.bashrc and source it)
    export PATH=/opt/openrobots/bin:$PATH
    export PKG_CONFIG_PATH=/opt/openrobots/lib/pkgconfig:$PKG_CONFIG_PATH
    export LD_LIBRARY_PATH=/opt/openrobots/lib:$LD_LIBRARY_PATH
    export PYTHONPATH=/opt/openrobots/lib/python2.7/site-packages:$PYTHONPATH
   Then, I started installing tsid following the instructions on this website and failed when executing cmake:  
[image]
  I am really confused.

olivier-stasse, 2019-10-12:
Dear @xuhuairuogu 
Several solutions are possible
1/ You want to continue with sources:
I would suggest to try:
```
export PKG_CONFIG_PATH=/opt/openrobots/lib/pkgconfig/:$PKG_CONFIG_PATH
```
Note that there is '/' just after pkgconfig
2/ You might try to use the binaries:
apt-get install robotpkg-tsid
apt-get install robotpkg-py27-tsid

Note that there is currently a problem with the python bindings, and I will push a PR to fix this problem this week-end.

xuhuairuogu, 2019-10-14:
Thanks for your reply. Now I have successfully compiled and installed pinocchio, OpenSceneGraph3.4, PythonQt, gepetto-viewer, gepetto-viewer-corba and tsid all from source. However, when I execute examples related to Gepetto GUI, for example exercizes/ex_1.py, 3D models cannot be displayed. It seems that everything except for model displaying works well.
[image]
[image]

jcarpent, 2019-10-14:
Hi @xuhuairuogu,

It seems that you did not install the osg-dae pluging necessary to load your meshes (because they are collada files).
You can do that using the following command:
`sudo apt-get install robotpkg-osg-dae`
See https://github.com/stack-of-tasks/pinocchio/issues/705 for additional details.

Best,

Justin

xuhuairuogu, 2019-10-15:
HI, on Ubuntu18.04, there is not robotpkg-osg-dae. Therefore, I install collada using the following command:
`sudo apt-get install libcollada-dom2.4-dp-dev`
and recompile OpenSceneGraph3.4 and osgdb_dea.so is generated.
However, when executing `python ex_1.py`, a new error appears:
[image]

xuhuairuogu, 2019-10-15:
I have converted all dae files to stl files by Blender and changed the 3D model file path in romeo.urdf. Although it is not an elegant way, it really works.

closed 2019-11-25 by andreadelprete