stack-of-tasks/tsid #53: Dependency to gepetto viewer corba.

olivier-stasse, 2019-10-15:
Would it be nice to have a dependency to the gepetto viewer, such that we can make sure that the appropriate package has been installed. If this is not relevant here we could do another meta package.
Best,
Olivier.

andreadelprete, 2019-10-16:
At a first glance I'd say it'd be better to have it in another meta package, because TSID per se doesn't need any viewer.

olivier-stasse, 2019-11-05:
My problem comes from this line which implies gepetto-viewer:
https://github.com/stack-of-tasks/tsid/blob/761194b6447e0380a1af8cff325b47e162af2f38/exercizes/ex_1.py#L73

andreadelprete, 2019-11-07:
I see. What about having it as optional dependency? Because one could still use TSID without the viewer.

commit 17d12e7b (2021-08-12): [exercises] update display only if we have a viewer, fix #53 (#140)