stack-of-tasks/tsid #58: TSID python bindings compilation error

DanWillans, 2019-11-08:
Hi,

I'm trying to build from source and without building python bindings everything is fine. When I try to build with python bindings I get an error. I am on Ubuntu 18.04.

> /usr/include/boost/python/object/py_function.hpp:36:15:   required from here
/usr/include/boost/variant/detail/make_variant_list.hpp:40:46: error: wrong number of template arguments (21, should be at least 0)
   40 |     typedef typename mpl::list< T... >::type type;
      |                                              ^~~~
In file included from /usr/include/boost/mpl/aux_/include_preprocessed.hpp:37,
                 from /usr/include/boost/mpl/list.hpp:46,
                 from /usr/include/boost/math/policies/policy.hpp:9,
                 from /usr/include/boost/math/special_functions/math_fwd.hpp:29,
                 from /usr/include/boost/math/special_functions/sign.hpp:17,
                 from /usr/include/boost/lexical_cast/detail/inf_nan.hpp:34,
                 from /usr/include/boost/lexical_cast/detail/converter_lexical_streams.hpp:63,
                 from /usr/include/boost/lexical_cast/detail/converter_lexical.hpp:54,
                 from /usr/include/boost/lexical_cast/try_lexical_convert.hpp:42,
                 from /usr/include/boost/lexical_cast.hpp:32,
                 from /usr/include/boost/python/operators.hpp:19,
                 from /usr/include/boost/python.hpp:49,
                 from /home/dan/Dev/tsid/_build-RELEASE/include/tsid/bindings/python/contacts/contact-6d.hpp:21,
                 from /home/dan/Dev/tsid/bindings/python/contacts/contact-6d.cpp:18:
/usr/include/boost/mpl/aux_/preprocessed/gcc/list.hpp:22:8: note: provided for ‘template<class T0, class T1, class T2, class T3, class T4, class T5, class T6, class T7, class T8, class T9, class T10, class T11, class T12, class T13, class T14, class T15, class T16, class T17, class T18, class T19> struct boost::mpl::list’
   22 | struct list;
      |        ^~~~
In file included from /usr/include/boost/variant.hpp:17,
                 from /usr/local/include/pinocchio/multibody/joint/joint-collection.hpp:11,
                 from /usr/local/include/pinocchio/multibody/joint/joint-generic.hpp:8,
                 from /usr/local/include/pinocchio/multibody/model.hpp:17,
                 from /home/dan/Dev/tsid/include/tsid/tasks/task-se3-equality.hpp:25,
                 from /home/dan/Dev/tsid/include/tsid/contacts/contact-6d.hpp:23,
                 from /home/dan/Dev/tsid/_build-RELEASE/include/tsid/bindings/python/contacts/contact-6d.hpp:26,
                 from /home/dan/Dev/tsid/bindings/python/contacts/contact-6d.cpp:18:
/usr/include/boost/variant/variant.hpp: In instantiation of ‘typename Visitor::result_type boost::variant<T0, TN>::internal_apply_visitor(Visitor&) [with Visitor = boost::detail::variant::destroyer; T0_ = pinocchio::JointDataRevoluteTpl<double, 0, 0>; TN = {pinocchio::JointDataRevoluteTpl<double, 0, 1>, pinocchio::JointDataRevoluteTpl<double, 0, 2>, pinocchio::JointDataMimic<pinocchio::JointDataRevoluteTpl<double, 0, 0> >, pinocchio::JointDataMimic<pinocchio::JointDataRevoluteTpl<double, 0, 1> >, pinocchio::JointDataMimic<pinocchio::JointDataRevoluteTpl<double, 0, 2> >, pinocchio::JointDataFreeFlyerTpl<double, 0>, pinocchio::JointDataPlanarTpl<double, 0>, pinocchio::JointDataRevoluteUnalignedTpl<double, 0>, pinocchio::JointDataSphericalTpl<double, 0>, pinocchio::JointDataSphericalZYXTpl<double, 0>, pinocchio::JointDataPrismaticTpl<double, 0, 0>, pinocchio::JointDataPrismaticTpl<double, 0, 1>, pinocchio::JointDataPrismaticTpl<double, 0, 2>, pinocchio::JointDataPrismaticUnalignedTpl<double, 0>, pinocchio::JointDataTranslationTpl<double, 0>, pinocchio::JointDataRevoluteUnboundedTpl<double, 0, 0>, pinocchio::JointDataRevoluteUnboundedTpl<double, 0, 1>, pinocchio::JointDataRevoluteUnboundedTpl<double, 0, 2>, pinocchio::JointDataRevoluteUnboundedUnalignedTpl<double, 0>, boost::recursive_wrapper<pinocchio::JointDataCompositeTpl<double, 0, pinocchio::JointCollectionDefaultTpl> >}; typename Visitor::result_type = void]’:
/usr/include/boost/variant/variant.hpp:1414:9:   required from ‘void boost::variant<T0, TN>::destroy_content() [with T0_ = pinocchio::JointDataRevoluteTpl<double, 0, 0>; TN = {pinocchio::JointDataRevoluteTpl<double, 0, 1>, pinocchio::JointDataRevoluteTpl<double, 0, 2>, pinocchio::JointDataMimic<pinocchio::JointDataRevoluteTpl<double, 0, 0> >, pinocchio::JointDataMimic<pinocchio::JointDataRevoluteTpl<double, 0, 1> >, pinocchio::JointDataMimic<pinocchio::JointDataRevoluteTpl<double, 0, 2> >, pinocchio::JointDataFreeFlyerTpl<double, 0>, pinocchio::JointDataPlanarTpl<double, 0>, pinocchio::JointDataRevoluteUnalignedTpl<double, 0>, pinocchio::JointDataSphericalTpl<double, 0>, pinocchio::JointDataSphericalZYXTpl<double, 0>, pinocchio::JointDataPrismaticTpl<double, 0, 0>, pinocchio::JointDataPrismaticTpl<double, 0, 1>, pinocchio::JointDataPrismaticTpl<double, 0, 2>, pinocchio::JointDataPrismaticUnalignedTpl<double, 0>, pinocchio::JointDataTranslationTpl<double, 0>, pinocchio::JointDataRevoluteUnboundedTpl<double, 0, 0>, pinocchio::JointDataRevoluteUnboundedTpl<double, 0, 1>, pinocchio::JointDataRevoluteUnboundedTpl<double, 0, 2>, pinocchio::JointDataRevoluteUnboundedUnalignedTpl<double, 0>, boost::recursive_wrapper<pinocchio::JointDataCompositeTpl<double, 0, pinocchio::JointCollectionDefaultTpl> >}]’
/usr/include/boost/variant/variant.hpp:1421:9:   required from ‘boost::variant<T0, TN>::~variant() [with T0_ = pinocchio::JointDataRevoluteTpl<double, 0, 0>; TN = {pinocchio::JointDataRevoluteTpl<double, 0, 1>, pinocchio::JointDataRevoluteTpl<double, 0, 2>, pinocchio::JointDataMimic<pinocchio::JointDataRevoluteTpl<double, 0, 0> >, pinocchio::JointDataMimic<pinocchio::JointDataRevoluteTpl<double, 0, 1> >, pinocchio::JointDataMimic<pinocchio::JointDataRevoluteTpl<double, 0, 2> >, pinocchio::JointDataFreeFlyerTpl<double, 0>, pinocchio::JointDataPlanarTpl<double, 0>, pinocchio::JointDataRevoluteUnalignedTpl<double, 0>, pinocchio::JointDataSphericalTpl<double, 0>, pinocchio::JointDataSphericalZYXTpl<double, 0>, pinocchio::JointDataPrismaticTpl<double, 0, 0>, pinocchio::JointDataPrismaticTpl<double, 0, 1>, pinocchio::JointDataPrismaticTpl<double, 0, 2>, pinocchio::JointDataPrismaticUnalignedTpl<double, 0>, pinocchio::JointDataTranslationTpl<double, 0>, pinocchio::JointDataRevoluteUnboundedTpl<double, 0, 0>, pinocchio::JointDataRevoluteUnboundedTpl<double, 0, 1>, pinocchio::JointDataRevoluteUnboundedTpl<double, 0, 2>, pinocchio::JointDataRevoluteUnboundedUnalignedTpl<double, 0>, boost::recursive_wrapper<pinocchio::JointDataCompositeTpl<double, 0, pinocchio::JointCollectionDefaultTpl> >}]’
/usr/local/include/pinocchio/multibody/joint/joint-generic.hpp:68:10:   required from ‘void __gnu_cxx::new_allocator<_Tp>::destroy(_Up*) [with _Up = pinocchio::JointDataTpl<double>; _Tp = pinocchio::JointDataTpl<double>]’
/usr/include/c++/9/bits/alloc_traits.h:260:4:   required from ‘static decltype (__a.destroy(__p)) std::allocator_traits<_Alloc>::_S_destroy(_Alloc2&, _Tp*, int) [with _Alloc2 = Eigen::aligned_allocator<pinocchio::JointDataTpl<double> >; _Tp = pinocchio::JointDataTpl<double>; _Alloc = Eigen::aligned_allocator<pinocchio::JointDataTpl<double> >; decltype (__a.destroy(__p)) = void]’
/usr/include/c++/9/bits/alloc_traits.h:364:14:   required from ‘static void std::allocator_traits<_Alloc>::destroy(_Alloc&, _Tp*) [with _Tp = pinocchio::JointDataTpl<double>; _Alloc = Eigen::aligned_allocator<pinocchio::JointDataTpl<double> >]’
/usr/include/c++/9/bits/stl_construct.h:198:19:   [ skipping 3 instantiation contexts, use -ftemplate-backtrace-limit=0 to disable ]
/usr/include/boost/python/detail/destroy.hpp:52:15:   required from ‘void boost::python::detail::destroy_referent_impl(void*, T& (*)()) [with T = const pinocchio::DataTpl<double>]’
/usr/include/boost/python/detail/destroy.hpp:58:26:   required from ‘void boost::python::detail::destroy_referent(void*, T (*)()) [with T = const pinocchio::DataTpl<double>&]’
/usr/include/boost/python/converter/rvalue_from_python_data.hpp:135:51:   required from ‘boost::python::converter::rvalue_from_python_data<T>::~rvalue_from_python_data() [with T = const pinocchio::DataTpl<double>&]’
/usr/include/boost/python/converter/arg_from_python.hpp:106:8:   required from ‘PyObject* boost::python::detail::caller_arity<5>::impl<F, Policies, Sig>::operator()(PyObject*, PyObject*) [with F = tsid::math::ConstraintInequality (*)(tsid::contacts::Contact6d [... cut ...]
                          robotpkg-pinocchio (= 1.2.4r3) but it is not going to be installed or
                          robotpkg-pinocchio (= 1.2.6r1) but it is not going to be installed
                 Depends: libboost-test1.58-dev (>= 1.34.1) but it is not installable
                 Depends: libboost-test1.58.0 (>= 1.34.1) but it is not installable
E: Unable to correct problems, you have held broken packages.

I have seen a similar issue on pinocchio github. https://github.com/stack-of-tasks/pinocchio/issues/837

Any help would be appreciated.

Thanks!

andreadelprete, 2019-11-08:
When you're building TSID from source, which version of Pinocchio are you using?

andreadelprete, 2019-11-08:
Moreover, which branch of TSID are you using? Master or devel?

andreadelprete, 2019-11-08:
I've just tried building from source the latest TSID (devel) with the latest pinocchio (devel) and it works (even with python bindings). The only difference is that I'm on 16.04.

DanWillans, 2019-11-08:
I just spoke with a colleague and we rebuilt using robotpkg to install pinocchio instead of from source. It seems to have solved all of the dependencies issues now and can rebuild TSID without issue. How strange. 

For any future people with the issue I followed this guide to install robotpkg http://robotpkg.openrobots.org/install.html

I will close the issue - thanks for looking @andreadelprete - Appreciate it.

closed 2019-11-08 by DanWillans