stack-of-tasks/tsid #59: MemoryError using Python bindings

WangKeAlchemist, 2019-11-21:
Hi,
I installed tsid and its python bindings successfully. But when I try to run the test script 'ipython script/test_formulation.py'. The python gives me the error 'MemoryError'. 
[image]
The same happens when I ran another test script:
[image]

I checked my memory and it has plenty. Also, I checked the version of Python, it is 64 bit. So I cannot figure out why...

BTW, when I want to switch to C++, I find very few examples and almost no tutorial, so I think python is the main language of the library, but any examples for C++?

WangKeAlchemist, 2019-11-21:
Also, I installed tsid from source. When I tried to install using the binary, using the command `sudo apt-get install robotpkg-tsid`. It showed that the conflict of dependency on pinocchio.
[image]

Basically I installed pinocchio with newest version while tsid depends on older ones, maybe you can resolve it :)

andreadelprete, 2019-11-25:
> Hi,
> I installed tsid and its python bindings successfully. But when I try to run the test script 'ipython script/test_formulation.py'. The python gives me the error 'MemoryError'.

Hi @WangKeAlchemist, I' ve tried running the script on my system with the last version of TSID (branch devel) and it works fine. Could you tell me which version of pinocchio you're using? 

The call to ```getNeutralConfigurationFromSrdf``` is deprecated in Pinocchio, and it is suggested to use ```loadReferenceConfigurations``` instead. The problem is that if I do so, it doesn't work (the function returns None). But maybe with the new version of Pinocchio (which I guess is the one you're using) it'll work. So can you try replacing that line with 
```
refConf = se3.loadReferenceConfigurations(robot.model(), srdf)
print refConf
```
and see what it prints? I guess refConf should be a dictionary, which you should be able to access with the name of the reference configuration you want.
 
> BTW, when I want to switch to C++, I find very few examples and almost no tutorial, so I think python is the main language of the library, but any examples for C++?

It's true that most examples are in python, because for education that's the best choice. However, TSID has been thought to be used in C++ for controlling real robots. [Here](https://github.com/stack-of-tasks/tsid/blob/master/unittest/tsid-formulation.cpp) you have some standalone examples. [Here](https://github.com/stack-of-tasks/sot-torque-control/blob/master/src/inverse-dynamics-balance-controller.cpp) you have an example showing how to use TSID for balancing a humanoid robot inside the dynamic-graph framework.

andreadelprete, 2019-11-25:
> Also, I installed tsid from source. When I tried to install using the binary, using the command `sudo apt-get install robotpkg-tsid`. It showed that the conflict of dependency on pinocchio.
> [image] [...]

For conflicts related to binary versions I would ask @nim65s what he thinks about this.

WangKeAlchemist, 2019-11-25:
> Hi @WangKeAlchemist, I' ve tried running the script on my system with the last version of TSID (branch devel) and it works fine. Could you tell me which version of pinocchio you're using?
>
> The call [...]

Hi @andreadelprete, thanks a lot for your detailed reply. I installed the newest version of Pinocchio, namely v2.1.9. And my tsid version is downloaded from git, from master branch.

I tried you suggestion by replacing with `refConf = se3.loadReferenceConfigurations(robot.model(), srdf)`, and I realize that the problem actually comes from `robot.model()` and it is THIS code that causes the error message. But my computer has 9GB free memory and 2GB free swap space, so te memory should not be the problem. Is it possible that there is a problem with the interface between tsid and pinocchio? I will try the newest devel branch code.

WangKeAlchemist, 2019-11-25:
Hi @andreadelprete , magically, I cleaned the old installation of pinocchio and tsid and re-installed, then it worked. Thanks a lot!

andreadelprete, 2019-11-26:
Glad to hear everything is working now! :)

closed 2019-11-26 by andreadelprete