stack-of-tasks/tsid #61: error ‘parseURDFFile’ is not a member of ‘urdf’

angelobratta, 2019-11-28:
Last week I have installed pinocchio through `sudo apt install robotpkg-py27-pinocchio` and now I want to install TSID.
I have followed the guide and when I try to install with `make install`, I get the following error when it is creating the `robot-wrapper.cpp.o`  

```
cd /opt/openrobots/src/tsid/_build-RELEASE/src && /usr/lib/ccache/c++   -DEIGEN_RUNTIME_NO_MALLOC -DROS_BUILD_SHARED_LIBS=1 -Dtsid_EXPORTS -I/opt/openrobots/src/tsid/_build-RELEASE -I/opt/openrobots/src/tsid/_build-RELEASE/include -I/opt/openrobots/src/tsid/include -isystem /usr/include/python2.7 -isystem /usr/include/eigen3 -isystem /opt/openrobots/include -isystem /usr/include/assimp  -pedantic -Wno-long-long -Wall -Wextra -Wcast-align -Wcast-qual -Wformat -Wwrite-strings -Wconversion  -O3 -DNDEBUG -fPIC   -DPINOCCHIO_URDFDOM_TYPEDEF_SHARED_PTR -DBOOST_MPL_LIMIT_LIST_SIZE=30 -DPINOCCHIO_WITH_URDFDOM -DPINOCCHIO_WITH_HPP_FCL -DFCL_USE_ASSIMP_UNIFIED_HEADER_NAMES -DHPP_FCL_USE_ASSIMP_UNIFIED_HEADER_NAMES -DHPP_FCL_HAVE_OCTOMAP -DFCL_HAVE_OCTOMAP -DOCTOMAP_MAJOR_VERSION=1 -DOCTOMAP_MINOR_VERSION=9 -DOCTOMAP_PATCH_VERSION=0 -o CMakeFiles/tsid.dir/robots/robot-wrapper.cpp.o -c /opt/openrobots/src/tsid/src/robots/robot-wrapper.cpp
In file included from /opt/openrobots/include/pinocchio/parsers/urdf.hpp:263:0,
                 from /opt/openrobots/src/tsid/src/robots/robot-wrapper.cpp:21:
/opt/openrobots/include/pinocchio/parsers/urdf/model.hxx: In function ‘pinocchio::ModelTpl<Scalar, Options, JointCollectionTpl>& pinocchio::urdf::buildModel(const string&, const typename pinocchio::ModelTpl<Scalar, Options, JointCollectionTpl>::JointModel&, pinocchio::ModelTpl<Scalar, Options, JointCollectionTpl>&, bool)’:
/opt/openrobots/include/pinocchio/parsers/urdf/model.hxx:618:50: error: ‘parseURDFFile’ is not a member of ‘urdf’
       ::urdf::ModelInterfaceSharedPtr urdfTree = ::urdf::parseURDFFile (filename);
                                                  ^
/opt/openrobots/include/pinocchio/parsers/urdf/model.hxx: In function ‘pinocchio::ModelTpl<Scalar, Options, JointCollectionTpl>& pinocchio::urdf::buildModel(const string&, pinocchio::ModelTpl<Scalar, Options, JointCollectionTpl>&, bool)’:
/opt/openrobots/include/pinocchio/parsers/urdf/model.hxx:638:50: error: ‘parseURDFFile’ is not a member of ‘urdf’
       ::urdf::ModelInterfaceSharedPtr urdfTree = ::urdf::parseURDFFile(filename);
                                                  ^
src/CMakeFiles/tsid.dir/build.make:785: recipe for target 'src/CMakeFiles/tsid.dir/robots/robot-wrapper.cpp.o' failed
```
How can I fix this?
Thanks

andreadelprete, 2019-11-28:
This looks like an internal problem of Pinocchio because the line that triggers the error (```robot-wrapper.cpp:21```) is simply including a file of pinocchio:
```#include <pinocchio/parsers/urdf.hpp>``` 
I've checked on the pinocchio repo and I couldn't find any open issue related to this. I'll keep investigating and let you know.

andreadelprete, 2019-11-28:
I've tried installing pinocchio from binaries as you did, and then I compiled the latest version of TSID (branch devel) and it works fine. Can you make sure that:
* you have the latest version of pinocchio (2.1.9): just execute ```dpkg -l | grep pinocchio```
* you have the latest version of TSID: execute ```git pull``` in the TSID repo
* you are using the devel branch of TSID: execute ```git checkout devel``` in the TSID repo
* you have the same version of urdfdom I have (0.4.1-1): if I run ```dpkg -l | grep urdf``` I get
```
ii  liburdfdom-dev:amd64                        0.4.1-1                                         amd64        URDF DOM - development files
ii  liburdfdom-headers-dev                      0.4.1-2                                         amd64        URDF DOM - header files
ii  liburdfdom-model-state0.4:amd64             0.4.1-1                                         amd64        URDF DOM - model state library
ii  liburdfdom-model0.4:amd64                   0.4.1-1                                         amd64        URDF DOM - model library
ii  liburdfdom-sensor0.4:amd64                  0.4.1-1                                         amd64        URDF DOM - sensor library
ii  liburdfdom-world0.4:amd64                   0.4.1-1                                         amd64        URDF DOM - world library
```
If none of these is the problem, could you copy-paste here the result of ```env```?

angelobratta, 2019-11-29:
I have the same versions. 
After the checkout and I got several errors when it is creating `robot-wrapper.cpp.o`

```
cd /opt/openrobots/src/tsid/_build-RELEASE/src && /usr/lib/ccache/c++   -DEIGEN_RUNTIME_NO_MALLOC -DROS_BUILD_SHARED_LIBS=1 -Dtsid_EXPORTS -I/opt/openrobots/src/tsid/_build-RELEASE -I/opt/openrobots/src/tsid/_build-RELEASE/include -I/opt/openrobots/src/tsid/include -isystem /usr/include/python2.7 -isystem /usr/include/eigen3 -isystem /opt/openrobots/include -isystem /usr/include/assimp  -pedantic -Wno-long-long -Wall -Wextra -Wcast-align -Wcast-qual -Wformat -Wwrite-strings -Wconversion  -O3 -DNDEBUG -fPIC   -DPINOCCHIO_URDFDOM_TYPEDEF_SHARED_PTR -DBOOST_MPL_LIMIT_LIST_SIZE=30 -DPINOCCHIO_WITH_URDFDOM -DPINOCCHIO_WITH_HPP_FCL -DFCL_USE_ASSIMP_UNIFIED_HEADER_NAMES -DHPP_FCL_USE_ASSIMP_UNIFIED_HEADER_NAMES -DHPP_FCL_HAVE_OCTOMAP -DFCL_HAVE_OCTOMAP -DOCTOMAP_MAJOR_VERSION=1 -DOCTOMAP_MINOR_VERSION=9 -DOCTOMAP_PATCH_VERSION=0 -o CMakeFiles/tsid.dir/robots/robot-wrapper.cpp.o -c /opt/openrobots/src/tsid/src/robots/robot-wrapper.cpp
In file included from /usr/local/include/urdf_model/joint.h:43:0,
                 from /usr/local/include/urdf_model/link.h:44,
                 from /usr/local/include/urdf_model/model.h:42,
                 from /opt/openrobots/include/pinocchio/parsers/urdf/types.hpp:8,
                 from /opt/openrobots/include/pinocchio/parsers/urdf.hpp:11,
                 from /opt/openrobots/src/tsid/src/robots/robot-wrapper.cpp:21:
/usr/local/include/urdf_model/pose.h: In member function ‘void urdf::Vector3::init(const string&)’:
/usr/local/include/urdf_model/pose.h:78:42: error: ‘to_string’ is not a member of ‘std’
       throw ParseError("Parser found " + std::to_string(xyz.size())  + " elements but 3 expected while parsing vector [" + vector_str + "]");
                                          ^
In file included from /usr/local/include/urdf_model/joint.h:44:0,
                 from /usr/local/include/urdf_model/link.h:44,
                 from /usr/local/include/urdf_model/model.h:42,
                 from /opt/openrobots/include/pinocchio/parsers/urdf/types.hpp:8,
                 from /opt/openrobots/include/pinocchio/parsers/urdf.hpp:11,
                 from /opt/openrobots/src/tsid/src/robots/robot-wrapper.cpp:21:
/usr/local/include/urdf_model/types.h: At global scope:
/usr/local/include/urdf_model/types.h:51:14: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
 typedef std::shared_ptr<double> DoubleSharedPtr;
              ^
/usr/local/include/urdf_model/types.h:53:25: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
 URDF_TYPEDEF_CLASS_POINTER(Box);
                         ^
/usr/local/include/urdf_model/types.h:53:68: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
/usr/local/include/urdf_model/types.h:53:122: error: ‘weak_ptr’ in namespace ‘std’ does not name a template type
In file included from /usr/local/include/urdf_model/joint.h:44:0,
                 from /usr/local/include/urdf_model/link.h:44,
                 from /usr/local/include/urdf_model/model.h:42,
                 from /opt/openrobots/include/pinocchio/parsers/urdf/types.hpp:8,
                 from /opt/openrobots/include/pinocchio/parsers/urdf.hpp:11,
                 from /opt/openrobots/src/tsid/src/robots/robot-wrapper.cpp:21:
/usr/local/include/urdf_model/types.h:54:31: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
 URDF_TYPEDEF_CLASS_POINTER(Collision);
                               ^
/usr/local/include/urdf_model/types.h:54:86: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
/usr/local/include/urdf_model/types.h:54:152: error: ‘weak_ptr’ in namespace ‘std’ does not name a template type
/usr/local/include/urdf_model/types.h:55:30: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
 URDF_TYPEDEF_CLASS_POINTER(Cylinder);
                              ^
/usr/local/include/urdf_model/types.h:55:83: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
/usr/local/include/urdf_model/types.h:55:147: error: ‘weak_ptr’ in namespace ‘std’ does not name a template type
/usr/local/include/urdf_model/types.h:56:30: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
 URDF_TYPEDEF_CLASS_POINTER(Geometry);
                              ^
/usr/local/include/urdf_model/types.h:56:83: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
/usr/local/include/urdf_model/types.h:56:147: error: ‘weak_ptr’ in namespace ‘std’ does not name a template type
/usr/local/include/urdf_model/types.h:57:30: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
 URDF_TYPEDEF_CLASS_POINTER(Inertial);
                              ^
/usr/local/include/urdf_model/types.h:57:83: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
/usr/local/include/urdf_model/types.h:57:147: error: ‘weak_ptr’ in namespace ‘std’ does not name a template type
/usr/local/include/urdf_model/types.h:58:27: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
 URDF_TYPEDEF_CLASS_POINTER(Joint);
                           ^
/usr/local/include/urdf_model/types.h:58:74: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
/usr/local/include/urdf_model/types.h:58:132: error: ‘weak_ptr’ in namespace ‘std’ does not name a template type
[... cut ...]
UPSTART_JOB=unity7
CMAKE_PREFIX_PATH=/home/abratta/dls_ws/install:/home/abratta/misc_ws/devel_isolated/towr_ros:/home/abratta/misc_ws/devel_isolated/towr:/home/abratta/misc_ws/devel_isolated/ifopt:/opt/ros/kinetic
XDG_RUNTIME_DIR=/run/user/1000
DISPLAY=:0
XDG_CURRENT_DESKTOP=Unity
GTK_IM_MODULE=ibus
LESSCLOSE=/usr/bin/lesspipe %s %s
LC_TIME=it_IT.UTF-8
ROS_ETC_DIR=/opt/ros/kinetic/etc/ros
LC_NAME=it_IT.UTF-8
XAUTHORITY=/home/abratta/.Xauthority
COLORTERM=gnome-terminal
_=/usr/bin/env
OLDPWD=/opt
```

andreadelprete, 2019-11-29:
It's weird that in your system urdfdom is installed in ```/usr/local/include```, while in mine it's in ```/usr/include```. Which OS are you using? I'm on Ubuntu 16.04.
Moreover, I've checked the content of the file ```urdf_model/pose.h``` and it's different from yours, so you must have a different version of urdfdom. Could you copy here the result of ```dpkg -l | grep urdf``` and ```dpkg -L liburdfdom-headers-dev```?

angelobratta, 2019-11-29:
I am on 16.04.6
urdf:
```
ii  liburdfdom-dev:amd64                         0.4.1-1                                               amd64        URDF DOM - development files
ii  liburdfdom-headers-dev                       0.4.1-2                                               amd64        URDF DOM - header files
ii  liburdfdom-model-state0.4:amd64              0.4.1-1                                               amd64        URDF DOM - model state library
ii  liburdfdom-model0.4:amd64                    0.4.1-1                                               amd64        URDF DOM - model library
ii  liburdfdom-sensor0.4:amd64                   0.4.1-1                                               amd64        URDF DOM - sensor library
ii  liburdfdom-tools                             0.4.1-1                                               amd64        URDF DOM - tools
ii  liburdfdom-world0.4:amd64                    0.4.1-1                                               amd64        URDF DOM - world library
ii  ros-kinetic-collada-urdf                     1.12.12-0xenial-20190608-042939-0800                  amd64        This package contains a tool to convert Unified Robot Description Format (URDF) documents into COLLAborative Design Activity (COLLADA) documents.
ii  ros-kinetic-urdf                             1.12.12-0xenial-20190608-034840-0800                  amd64        This package contains a C++ parser for the Unified Robot Description Format (URDF), which is an XML format for representing a robot model.
ii  ros-kinetic-urdf-parser-plugin               1.12.12-0xenial-20190607-172532-0800                  amd64        This package contains a C++ base class for URDF parsers.
ii  ros-kinetic-urdf-tutorial                    0.3.0-1xenial-20190608-132116-0800                    amd64        This package contains a number of URDF tutorials.
```
liburdfdom-headers-dev:

```
/.
/usr
/usr/lib
/usr/lib/pkgconfig
/usr/lib/pkgconfig/urdfdom_headers.pc
/usr/include
/usr/include/urdf_model
/usr/include/urdf_model/model.h
/usr/include/urdf_model/joint.h
/usr/include/urdf_model/types.h
/usr/include/urdf_model/color.h
/usr/include/urdf_model/twist.h
/usr/include/urdf_model/link.h
/usr/include/urdf_model/pose.h
/usr/include/urdf_exception
/usr/include/urdf_exception/exception.h
/usr/include/urdf_world
/usr/include/urdf_world/types.h
/usr/include/urdf_world/world.h
/usr/include/urdf_sensor
/usr/include/urdf_sensor/types.h
/usr/include/urdf_sensor/sensor.h
/usr/include/urdf_model_state
/usr/include/urdf_model_state/types.h
/usr/include/urdf_model_state/twist.h
/usr/include/urdf_model_state/model_state.h
/usr/share
/usr/share/urdfdom_headers
/usr/share/urdfdom_headers/cmake
/usr/share/urdfdom_headers/cmake/urdfdom_headers-config-version.cmake
/usr/share/urdfdom_headers/cmake/urdfdom_headers-config.cmake
/usr/share/doc
/usr/share/doc/liburdfdom-headers-dev
/usr/share/doc/liburdfdom-headers-dev/copyright
/usr/share/doc/liburdfdom-headers-dev/changelog.Debian.gz
/usr/share/doc/liburdfdom-headers-dev/README.md
```

andreadelprete, 2019-11-29:
Ok, the problem is that you have two versions of urdfdom (ros-kinetic-urdf and liburdfdom), and TSID is using the wrong one (ros) to compile, which is not compatible with pinocchio. There are many ways to fix this. If you don't need it, you can simply uninstall ros-kinetic-urdf. Otherwise if you need to keep both, you can simply change you PKG_CONFIG_PATH so that the correct library is found first (i.e. add ```/usr/lib/x86_64-linux-gnu/pkgconfig``` at the beginning). 

To see where pkg-config (version 0.24 or later) searches for installed libraries by default, use the following command:
```
pkg-config --variable pc_path pkg-config
```

angelobratta, 2019-11-29:
I need both, so I have modified the PKG_CONFIG_PATH as you can see here:
```
usr/lib/x86_64-linux-gnu/pkgconfig:/opt/openrobots/lib/pkgconfig:/home/abratta/dls_ws/install/lib/pkgconfig:/home/abratta/misc_ws/devel_isolated/towr_ros/lib/pkgconfig:/opt/ros/kinetic/lib/pkgconfig:/opt/ros/kinetic/lib/x86_64-linux-gnu/pkgconfig:/usr/local/lib/pkgconfig

```
I got the same errors.
The output of `pkg-config --variable pc_path pkg-config` is:
```
/usr/local/lib/x86_64-linux-gnu/pkgconfig:/usr/local/lib/pkgconfig:/usr/local/share/pkgconfig:/usr/lib/x86_64-linux-gnu/pkgconfig:/usr/lib/pkgconfig:/usr/share/pkgconfig
```

andreadelprete, 2019-11-29:
Have you done a clean build of TSID (i.e. removed all files from the build folder)?
If so, it is possible that the other library is still found first by pkg-config. You should configure your system so that the right library is found first. I think that online you can find good tutorials/documentation explaining how pkg-config works.

angelobratta, 2019-12-02:
I will look for it online.
Thank you

closed 2020-01-20 by andreadelprete