stack-of-tasks/tsid #62: Add position bounds on the ODRI quadruped

sduminy, 2019-12-11:
Hi,
I would like to add position bounds to the knees of the quadruped to prevent the legs to turn around themselves. 
I would like to know where the position bounds are taken into account in tsid ?
I have tried to modify a copy of the robot.model.lowerPositionLimits and the robot.model.upperPositionLimits and then set it as new model when using tsid but the new boundaries were not taken into account.
I have also thought of adding a new task using TaskJointBounds but we can not set Position bounds with this task (only velocity or acceleration bounds).
If you have an answer or some leads I could look into, that would be amazing.
Thanks for your advices.

andreadelprete, 2019-12-11:
Hi @sduminy , currently joint position bounds are not implemented. If you want, I could give you some tips on how to implement them in TaskJointBounds. As a starting point you could take a look at this paper [1], which explains the theory.

[1] https://hal.archives-ouvertes.fr/hal-01356989v3/document

andreadelprete, 2019-12-11:
And [here](https://github.com/andreadelprete/pinocchio_inv_dyn/blob/master/python/pinocchio_inv_dyn/acc_bounds_util.py) you can find a python implementation of the method described in the paper I mentioned above.

sduminy, 2019-12-11:
Okay, I will look into it and come back when I would have found a solution.
Thanks.

sduminy, 2019-12-13:
I will first add an emergency stop when position limits aren't respected to test it on the robot, so I won't implement the position bounds in TaskJointBounds. I will close this issue for now and maybe reopen it later if I have time to tackle this.
Thank you for your answer and advices !

dalinel, 2020-03-26:
Hello,
In order to implement joint position limits for Talos I have looked at the article and python code

I have implemented the C++ version of the python code inside tsid

The joint position limit seem to work 

I need to do more testing , to test other velocity and acceleration limits as well as check the generated torque. I will keep you updated on this 

Then it would be interesting to combine it with torque limits too

andreadelprete, 2020-03-26:
Hi @dalinel 
thanks for the update and for contributing to this project! When you're ready you can do a pull request so I can review the code.
Andrea

andreadelprete, 2020-05-09:
Since the PR has been merged I think we can close this issue.

closed 2020-05-09 by andreadelprete