stack-of-tasks/tsid #63: Problem referencing Pinocchio

mklvcm, 2020-01-17:
Hi guys,

I'm trying to follow your installation instructions and I'm getting the following error:

[image]

Any ideas what can be wrong?

Thanks,
Misha.

andreadelprete, 2020-01-17:
Probably that method was deprecated and now it has been removed from pinocchio. Maybe you can try replacing it with:
```
q = se3.getNeutralConfiguration(robot.model(), srdf, False)
```
or
```
se3.loadReferenceConfigurations(robot.model(), srdf, False)
q = robot.model().referenceConfigurations["name_of_the_configuration"]
```
where you should replace ```name_of_the_configuration``` with the actual name of the configuration you want to use (which should be written in the srdf file).

mklvcm, 2020-01-18:
Thank you Andrea.

Yes, the method seems deprecated. The method you suggested does not seem to be there neither.

I also tried to build **pinocchio** v2.10 from March last year. That version has the function. But when I run it it crashes with core dump :(

I'd like to give this library a try but I'm kind of stuck. Do you know of any versions that are proved working? Do you by any chance have a VM with all stuff installed? Or something?..

Thanks,
Misha.

andreadelprete, 2020-01-19:
Hi Misha,

I' ve just pushed in the devel branch a fix to your problem. I've also converted all python files to python 3 (pinocchio has already switched, so it was about time TSID did it too). For playing with the library I suggest you to use the scripts in the folder exercises (rather than the ones in tests).

mklvcm, 2020-01-20:
That is massively appreciated! I'll tell all my friends that Andrea is the man! Let me give it a try now .. :)

closed 2020-01-20 by andreadelprete