stack-of-tasks/tsid #65: [Cmake] catkin_package required for installation

paLeziart, 2020-01-29:
Dear all,

I ran into an issue trying to install TSID. It is asking for catkin-pkg but this package is not in the dependencies of [the main page.](https://github.com/stack-of-tasks/tsid#dependencies)

It seems there is some discussion going on about catkin preventing packaging in #48 involving the following lines in CMakeList.txt.
```
# Fail-safe support for catkin-ized pinocchio:
#  - If catkin-based pinocchio is installed it runs the CFG_EXTRAS to set up the Pinocchio preprocessor directives
#  - If it isn't, nothing happens and the subsequent pkg-config check takes care of everything.
find_package(catkin QUIET COMPONENTS pinocchio)
```

Looks like it is not supposed to block me if I do not have catkin_pkg.

Here is the output of my terminal:
```cmake .. -DCMAKE_BUILD_TYPE=RELEASE -DCMAKE_INSTALL_PREFIX=/local/users/paleziart -DPYTHON_EXECUTABLE=$(which python3.5) -DPYTHON_STANDARD_LAYOUT=ON
-- Option EIGEN_RUNTIME_NO_MALLOC on.
-- eigen3 >= 3.2.0 is required.
-- Checking for module 'eigen3 >= 3.2.0'
--   Found eigen3 , version 3.2.92
-- Pkg-config module eigen3 v3.2.92 has been detected with success.
-- Using CATKIN_DEVEL_PREFIX: /local/users/paleziart/tsid/_build-RELEASE/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/kinetic
-- This workspace overlays: /opt/ros/kinetic
-- Found PythonInterp: /usr/bin/python3.5 (found suitable version "3.5.2", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3.5
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /local/users/paleziart/tsid/_build-RELEASE/test_results
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found PythonInterp: /usr/bin/python3.5 (found version "3.5.2") 
-- Using Python nosetests: /usr/bin/nosetests
ImportError: "from catkin_pkg.package import parse_package" failed: No module named 'catkin_pkg'
Make sure that you have installed "catkin_pkg", it is up to date and on the PYTHONPATH.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  execute_process(/usr/bin/python3.5
  "/opt/ros/kinetic/share/catkin/cmake/parse_package_xml.py"
  "/opt/ros/kinetic/share/catkin/cmake/../package.xml"
  "/local/users/paleziart/tsid/_build-RELEASE/catkin/catkin_generated/version/package.cmake")
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package_xml.cmake:74 (safe_execute_process)
  /opt/ros/kinetic/share/catkin/cmake/all.cmake:168 (_catkin_package_xml)
  /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:80 (find_package)

-- Configuring incomplete, errors occurred!
See also "/local/users/paleziart/tsid/_build-RELEASE/CMakeFiles/CMakeOutput.log".
See also "/local/users/paleziart/tsid/_build-RELEASE/CMakeFiles/CMakeError.log".
```

nim65s, 2020-01-29:
This is the issue I always have with this line, I should have been more explicit when I tried to remove it in  https://github.com/stack-of-tasks/tsid/commit/50144111d09f722e01b0fbccdc37b91b6630d9e3

I guess that parts of catkin are found, and therefore it assumes that everything is available and functionnal, while it is not. Here, we want to build with python 3, and packages from the ROS binary package repositories are not available in python 3, so catkin is found, but the python modules are not.

I think we could remove this line, and use another way to package TSID on ROS ?

andreadelprete, 2020-01-31:
Sounds good to me!

nim65s, 2020-03-30:
Fixed in 1.2.3.
For use with ROS (at least catkin build), #75 should do the job. If there is anyother need, please re-open :)

closed 2020-03-30 by nim65s