stack-of-tasks/tsid #66: Python binding of RobotWrapper for devel branch

paLeziart, 2020-02-04:
Dear all,

I am working on the devel branch to get the recent commits related to Python 3 compatibility. I ran into an issue with Python binding for tsid.RobotWrapper. This issue has already been raised in #50 and corrected by the PR #51 but only for the master branch so you will likely get this error when testing the scripts on the devel branch.

A quick solution for that if you are on devel is `git cherry-pick 761194b`. It is just to warn you if you have this issue in the future.

Best,

nim65s, 2020-02-12:
The good solution would be to commit only on the devel branch.

master should only get merges. I'll fix that, by making a new release.

nim65s, 2020-03-01:
done, sorry for the delay.

closed 2020-03-01 by nim65s