stack-of-tasks/tsid #67: [Feature] Task force is not implemented

thomasfla, 2020-02-05:
from https://github.com/stack-of-tasks/tsid/blob/761194b6447e0380a1af8cff325b47e162af2f38/src/tasks/task-contact-force.cpp#L1 It looks like the task force in not implemented yet.
@andreadelprete, any advise on how to implement it? Or a workaround to set a pure force task for a 3d contact?

andreadelprete, 2020-02-05:
You can simply use the methods [setForceReference](https://github.com/stack-of-tasks/tsid/blob/761194b6447e0380a1af8cff325b47e162af2f38/include/tsid/contacts/contact-point.hpp#L100) and [setRegularizationTaskWeightVector](https://github.com/stack-of-tasks/tsid/blob/761194b6447e0380a1af8cff325b47e162af2f38/include/tsid/contacts/contact-point.hpp#L101) of `ContactPoint`.

thomasfla, 2020-02-05:
Thank you very much @andreadelprete !

closed 2020-02-05 by andreadelprete