stack-of-tasks/tsid #70: Wrong convergence of the contact force reference

thomasfla, 2020-02-20:
@paleziart repported a wrong convergence of the contact forces when a reference is given.

The contact force equality task seems to be miss-formulated :
https://github.com/stack-of-tasks/tsid/blob/761194b6447e0380a1af8cff325b47e162af2f38/src/contacts/contact-point.cpp#L109

I think `m_forceRegTask.setVector(m_fRef);` should be replaced by `m_forceRegTask.setVector(m_fRef * m_weightForceRegTask);`

Same applies https://github.com/stack-of-tasks/tsid/blob/761194b6447e0380a1af8cff325b47e162af2f38/src/contacts/contact-6d.cpp#L141

andreadelprete, 2020-02-20:
You're right, that's a mistake! Probably nobody noticed before because typically the reference force is zero. It should be
```
m_forceRegTask.setVector(A*m_fRef);
```
Could you fix that? Otherwise I can try to do it tomorrow.

thomasfla, 2020-02-20:
`m_forceRegTask.setVector(m_fRef * m_weightForceRegTask);` right?
I'll PR the fix. Thanks !

andreadelprete, 2020-02-21:
> `m_forceRegTask.setVector(m_fRef * m_weightForceRegTask);` right?

Not sure about this. I am sure it'd work using the matrix `A`, because it's a classic matrix-vector product, where `A` is a diagonal matrix, whose diagonal is `m_weightForceRegTask`.
If instead you multiply `m_weightForceRegTask * m_fRef` you're multiplying two vectors of the same size, and I'm afraid it's gonna compute their dot product.

andreadelprete, 2020-03-28:
Hi @thomasfla 
any news on this issue? If it's solved, could you close it?

thomasfla, 2020-05-13:
Thanks @paLeziart

andreadelprete, 2020-05-13:
Thanks guys!

closed 2020-05-13 by andreadelprete