stack-of-tasks/tsid #73: [task-se3-equality] Jdot_qdot term

PedroDesRobots, 2020-03-27:
Hi @andreadelprete ,

I want to implement a Cartesian constrain inside TSID, to be able later to perform a constraint on the convex hull (a cartesian constraint on CoM). 

I have taken a look at the Cartesian task: task-se3-equality
And regarding the current implementation, I got a question concerning how the matrix A and vector b are computed.

        m_constraint.matrix().row(idx) = m_J.row(i);
        m_constraint.vector().row(idx) = (m_a_des - m_drift.toVector()).row(i);

Why are you not computing the term Jdot_qdot concerning the acceleration? I guess I missed something.

andreadelprete, 2020-03-27:
Hi @PedroDesRobots ,
`m_drift` contains exactly `Jdot_q_dot`. Does that answer your question?
Andrea

PedroDesRobots, 2020-03-28:
Thank you for your prompt answer.
I was thinking that `m_drift` was related to the cartesian acceleration. 
To be sure,  `m_drift` is computed with the function `frameClassicAcceleration`, which call

```
      const Frame & f = m_model.frames[index];
      frameAcceleration = f.placement.actInv(data.a[f.parent]);
      Motion v = f.placement.actInv(data.v[f.parent]);
      frameAcceleration.linear() += v.angular().cross(v.linear());
```

Are those lines computed the Jdot_qdot? Did you know some literature where I could learn more about that?

andreadelprete, 2020-03-28:
This is indeed a trick that's not very well documented in the literature, but it's very simple to explain. We know that the Cartesian acceleration ddx is related to the joint accelerations via the equation:
```
ddx = J*ddq + dJ*dq
```
We also know that we can compute ddx using the forward phase of the Recursive Newton Euler algorithm (RNEA). So, rather than trying to compute dJ, and then multiplying it times dq, we could directly compute ddx using RNEA setting ddq=0. Is that clear now?

PedroDesRobots, 2020-03-28:
Perfect, thank you very much for taking the time to explain this trick!

closed 2020-03-28 by PedroDesRobots

andreadelprete, 2020-03-28:
No problem!