stack-of-tasks/tsid #77: angular-momentum-equality task is not using ref.pos

pFernbach, 2020-04-01:
The task-angular-momentum-equality take the reference momentum from the `trajectorySample.vel()` value (https://github.com/stack-of-tasks/tsid/blob/master/src/tasks/task-angular-momentum-equality.cpp#L98) and it's derivative from the `trajectorySample.acc()` value (https://github.com/stack-of-tasks/tsid/blob/master/src/tasks/task-angular-momentum-equality.cpp#L103).

I think that I understand the logic here that a momentum is relative to a 'velocity'. But as the name of the task suggest I would have expected to set the momentum reference from the `trajectorySample.pos()` value and it's derivative from the `trajectorySample.vel()` value. 

Can I propose a PR to change this or should we keep the current behavior ?

andreadelprete, 2020-04-01:
I don't have a strong opinion on this, so I'd tend to agree with you, unless somebody thinks this is a good idea. I suggest to wait a couple of days to make sure nobody is against this for some good reasons, and then you can change it.

jcarpent, 2020-04-09:
I think the naming pos, vel and acc are misleading in fact. Another naming would be expected here I would say.

andreadelprete, 2020-04-09:
Maybe we could replace `vel` with `rate_of_change`? But what could we use to replace `pos`? And `acc`?

pFernbach, 2020-04-09:
It could be `value`, `time_derivative`,
 `time_second_derivative` ?

andreadelprete, 2020-04-09:
I like `value`! Maybe I'd go for `first_time_derivative` and `second_time_derivative` though.

jcarpent, 2020-04-10:
Yes, I do agree that `value` seems more appropriete.

andreadelprete, 2020-07-10:
@pFernbach Do you plan to do the PR? Or should I do it instead?

pFernbach, 2020-07-10:
Sorry I forgot about, I'll do it next week. Do you confirm the following changes:
* rename `TrajectorySample::pos, vel, acc` to `value, time_derivative, second_time_derivative`
* Adapt all the C++ code
* Set the current python API with `pos, vel, acc` deprecated and add a new one with `value, time_derivative, second_time_derivative`

andreadelprete, 2020-07-10:
Yes, that was the plan, but now that I think about it, since a trajectory is a function of time only, maybe we don't need to specify that the derivatives are wrt time. Therefore, we could simplify the names to `value, derivative, second_derivative`. What do you think?

pFernbach, 2020-07-10:
Yes it make sense, I don't think that there could be any confusion.

andreadelprete, 2021-08-21:
Shall we close this now?

commit ff427110 (2021-08-21): Refactor TrajectorySample::pos, vel, acc to value, derivative, second_derivative (#101)

* Refactor TrajectorySample::pos, vel, acc to value, derivative, second_derivative

* [Tests] update tests to refactor of TrajectorySample

* Task-am-equality: correctly use reference value/derivative instead of derivative/second_derivative
Fix #77

* [Python] update python bindings of TrajectorySample

* [Python] deprecate TrajectorySample.{pos,vel,acc}

* [Tests] python < 3.2 doesn't have assertWarns

* trajectory: keep deprecated pos/vel/acc, add updated getters / setters instead

* [CMake] install macr [... cut]
closed 2021-10-19 by jcarpent