stack-of-tasks/tsid #79: self collision avoidance

dalinel, 2020-04-08:
Hi,
Do you have any ideas or tips about how to implement self collision avoidance for tsid ?
Thank you in advance

andreadelprete, 2020-04-08:
Good question, but I don't have a good answer unfortunately. This is a difficult problem, at least in theory, and I don't have much experience with that in practice. You can try to see it as joint limits that act in Cartesian space. If you know the position, velocity and acceleration limits of a given point on the robot, then you can use the same viability analysis of my RAL-2018 paper to bound the current Cartesian acceleration of that point in order to avoid a future collision.

The problem with this approach is of course the initial assumption of knowing the acceleration limits, which should also be constant if you wanna use the same math of RAL-2018. In practice we know that acceleration limits depend on the robot configuration, so they change over time. You can try to be conservative and set rather small acceleration bounds, so that you never approach collision with a high velocity, increasing your chances of collision avoidance. In theory, this is all rather heuristic. In practice it could work well. You may wanna try asking people who have played with this problem more than me. For instance Hilario Tome (ex PAL robotics), or Joris Vaillant.

Finally, of course, take a look at the literature, for instance
[1] Rodríguez-Seda, E. J., Stipanović, D. M., & Spong, M. W. (2015). Guaranteed Collision Avoidance for Autonomous Systems with Acceleration Constraints and Sensing Uncertainties. Journal of Optimization Theory and Applications, (November). https://doi.org/10.1007/s10957-015-0824-7

dalinel, 2020-04-09:
Ok thanks for the insight !
I will look at that in detail

olivier-stasse, 2020-07-22:
If you have a fast algorithm to compute a body-to-body distance you can create a repulsive vector field.
The shape that you use may have an impact on the velocity continuity.

There is an example here:
Real-time (self)-collision avoidance task on a hrp-2 humanoid robot
O Stasse, A Escande, N Mansard, S Miossec, P Evrard, A Kheddar
2008 ieee international conference on robotics and automation, 3200-3205

If you are not too worried about reducing the workspace it is possible to use capsule (aka sweeping spheres)

andreadelprete, 2021-03-08:
I close the issue as it has been inactive for a while now.

dalinel, 2021-03-09:
Hi,
@jbmouret and @Timothee-ANNE have developed a task which allows to create repulsors between a tracked frame and frames to avoid
We are using it to prevent the gripper frame to collide with other body frames
It is here :
https://github.com/resibots/inria_wbc/blob/devel/src/tsid/task-self-collision.cpp
Tell us if you are interested in integrating it to tsid directly

andreadelprete, 2021-03-09:
That's interesting. Personally I think it could be useful for other TSID's users, so it would be great to integrate it in TSID.

jcarpent, 2021-03-10:
I do agree with @andreadelprete for the general use case of such a feature.
I would also add that Thibault Noël has also been working on such a feature (https://hal.laas.fr/hal-03000951/document). I would suggest having a more complete discussion to see how these two approaches can be merged in some sense to give rise to a unified API.
I let @nmansard comment further.

andreadelprete, 2022-01-26:
Any news on this subject?