stack-of-tasks/tsid #80: error from demo_romeo.py

TaeilJin, 2020-04-27:
Hi, 
I am trying to use TSID for generating whole body quadruped motion.
After installing the TSID, I run the demo_romeo.py.

There was a error message which can see in below.
[image]

It seems that the version of pinocchio is not updated, so I change "getNeutralConfiguration" to "neutral" function. 
After that, there was also another error message.
[image]

Could you tell me how to fix it?

andreadelprete, 2020-04-27:
Hi @TaeilJin 
first of all I'd need some more information. Which branch of TSID are you using? master or devel? Or did you install it with robotpkg? 
Which version of python are you using? 2 or 3? On which OS are you? Which version of pinocchio are you using?

TaeilJin, 2020-04-28:
Thank you for your quick response!

I use python 2.7 in Linux (ubuntu 18.04).
 
I build TSID from source of master branch.
There was no src folder in /opt/openrobots, so I cloned git(master branch) on my /home/myOwn folder and then modified the line " -DCMAKE_INSTALL_PREFIX=$**"opt"**/openrobots "

I installed the pinocchio using robotpkg. Based on my knowledge, I can't find the version of the pinocchio.

olivier-stasse, 2020-04-28:
To get pinocchio version could you tell us the result of:
```
dpkg -l | grep robotpkg-pinocchio
```
The third column will tell us the version of the package.

TaeilJin, 2020-04-28:
Thank you for instruction!

It tells version of pinocchio is 2.3.1

andreadelprete, 2020-04-30:
Hi @TaeilJin 
sorry for the delay, hard period for me in Italy.

At a first glance the problem seems related to mixing numpy arrays and matrices. Probably you're using a version of pinocchio that by default uses arrays (which is the new standard), but that script is still using matrices. You may fix that quickly by asking pinocchio (or more precisely eigenpy) to use matrices. The command to do so is
```
import eigenpy
eigenpy.switchToNumpyMatrix()
```

Moreover, it seems you're using one of the few scripts that has not been converted yet from python 2 to python 3. You may wanna switch to using python 3, since python 2 won't be supported any more (eventually).

andreadelprete, 2020-05-09:
Hi @TaeilJin 
I've just pushed on the devel branch of TSID an updated version of demo_romeo.py, which works with numpy.array, so it should be compatible with pinocchio and eigenpy now.
I've also upgraded it to Python 3, which now is the new standard, so I suggest you to switch to Python 3 as well. This should solve the issue, so I close it. Feel free to re-open it if you have some more questions.

closed 2020-05-09 by andreadelprete