stack-of-tasks/tsid #90: Frames handling for task-se3-equality

paLeziart, 2020-05-13:
Dear all,

Recently I checked the code of the task-se3-equality that I am using for feet in swing phase and I want to be sure I am not misunderstanding something.

In the compute function of task-se3-equality (https://github.com/stack-of-tasks/tsid/blob/devel/src/tasks/task-se3-equality.cpp#L164), you first create a SE3 object called oMi that you initialize with `m_robot.framePosition(data, m_frame_id, oMi);`. 

In the function framePosition (https://github.com/stack-of-tasks/tsid/blob/devel/src/robots/robot-wrapper.cpp#L207), the placement of the frame is retrieved like that `framePosition = data.oMi[f.parent].act(f.placement);` with `data.oMi[f.parent]` being the vector of absolute joint placements (wrt the world) according to Pinocchio documentation and `f.placement` being the placement of the frame wrt the parent joint. So from what I understand and with the numerical values I've read framePosition contains the current placement of the foot frame wrt the world.

The desired placement of the foot frame is set with the setReference function (https://github.com/stack-of-tasks/tsid/blob/devel/src/tasks/task-se3-equality.cpp#L96) with `vectorToSE3(ref.pos, m_M_ref);`. I'm setting this desired placement wrt world frame.

The error in position for the task is then `errorInSE3(oMi, m_M_ref, m_p_error);          // pos err in local frame` with the errorInSE3 function doing `error = pinocchio::log6(Mdes.inverse() * M);`. Since both the current and desired placements are in world frame, this error is too.

So in the rest of the `compute` function if we are working in local frame (`m_local_frame = True`) then this position error is directly mixed with other quantities expressed in the local frame of the foot. 
```
 // desired acc in local frame
 m_a_des = - m_Kp.cwiseProduct(m_p_error_vec)
                  - m_Kd.cwiseProduct(m_v_error.toVector())
                  + m_wMl.actInv(m_a_ref).toVector();
```
And if we are not working in local frame (`m_local_frame = False`) this position error is transformed from local to world frame before being used, except it is already in world frame.
`m_p_error_vec = m_wMl.toActionMatrix() * m_p_error.toVector();  // pos err in local world-oriented frame`

Could you tell me if I misunderstood something? Maybe the reference position of the foot should be given in its local frame? Thanks a lot!

Best regards,

andreadelprete, 2020-05-13:
> The error in position for the task is then `errorInSE3(oMi, m_M_ref, m_p_error); // pos err in local frame` with the errorInSE3 function doing `error = pinocchio::log6(Mdes.inverse() * M);`. Since both the current and desired placements are in world frame, this error is too.

Mdes and M are Homogenous transformations from local to world frame. However, the product `Mdes.inverse() * M` is a transformation from real local to desired local frame, thus `errorInSE3` gives you the error in local frame.

paLeziart, 2020-05-13:
Oh, I see. Thank you for your explanation @andreadelprete !

closed 2020-05-13 by paLeziart