stack-of-tasks/tsid #93: "print" in function solve

ABonnefoy, 2020-05-28:
Since the last package update, when doing 

```python
solver = tsid.SolverHQuadProgFast("qp solver")
solver.resize(invdyn.nVar, invdyn.nEq, invdyn.nIn) 
sol = solver.solve(HQPData)
```

a lot of data are printed, like for example 

```python
Unconstrained solution: -1.01038
Add constraint 0/1
Add constraint 1/2
Add constraint 2/3
Add constraint 3/4
Add constraint 4/5
Add constraint 5/6
Add constraint 6/7
Add constraint 7/8
Add constraint 8/9
Add constraint 9/10
Add constraint 10/11
Add constraint 11/12
Add constraint 12/13
Add constraint 13/14
Add constraint 14/15
Trying with constraint 6
Step direction z
Step sizes: 2.87851e-05 (t1 = inf, t2 = 2.87851e-05)  in both spaces: 0.000121391
Full step has taken 2.87851e-05
Add constraint 15/16
Trying with constraint 7
Step direction z
Step sizes: 1.53276e-05 (t1 = 7.59971e-05, t2 = 1.53276e-05)  in both spaces: 0.000126826
Full step has taken 1.53276e-05
Add constraint 16/17
```

It spams a lot, is there any way to 
* either do the printing in a separate function (like `HQPData.print_all()` for example)
* either adding an argument to remove the printing

Thanks

nim65s, 2020-05-28:
Hi @ABonnefoy and thanks for this issue.

I think this is coming from the recent change to https://github.com/stack-of-tasks/eiquadprog in #87 . I'll have a look at this issue

ABonnefoy, 2020-05-28:
Ok, thank you !

nim65s, 2020-05-28:
A definitive solution is on its way. 

If you want to remove the spam before it lands in your binary installation, you can redirect stderr to somewhere else, with eg. 
```
python my_script.py 2> /tmp/my_script.err
```

You will still get the standard message stream, but this flood will be redirected to the `/tmp/my_script.err`file.

ABonnefoy, 2020-05-28:
Great, thank you !

closed 2020-05-28 by ABonnefoy