stack-of-tasks/tsid #98: Generalize RobotWrapper.position and RobotWrapper.framePosition

pFernbach, 2020-07-08:
Most of the demo/exercizes are writted assuming the "feet names" represent joints in the pinocchio model (see eg. https://github.com/stack-of-tasks/tsid/blob/master/demo/demo_romeo.py#L89, also the name in this script are misleading `rf_frame_name` is the name of a joint in the pinocchio model). However it is not always true, like for Talos where the feet are represented by frames. 

This may cause confusion or create hard to find issue when using this scripts and changing robot model because the call to `robot.position(data, robot.model().getJointId(ee_name))` still execute without error even if `ee_name` is the name of a frame, but in this case it return garbage. 

I had to add the following method in my code to be able to deal with the fact that `ee_name` could be a joint or a frame:

```
def get_current_effector_position(robot, data, ee_name):
    id = robot.model().getJointId(ee_name)
    if id < len(data.oMi):
        return robot.position(data, id)
    else:
        id = robot.model().getFrameId(ee_name)
        return robot.framePosition(data, id)
```

I do not know what is your policy for checking user inputs when calling RobotWrapper methods, but I think that `RobotWrapper.position()` should check that the id is valid. (Either with an assert, warning message or throw an error as you prefer).

andreadelprete, 2020-07-09:
Agreed. Actually that's a bug that has been propagated through the scripts. My policy is to always use frames and never use joints, because anyway there's a frame associated to each joint, so there's no reason for not using frames. The piece of code you reported should actually be:
```
robot.position(data, robot.model().getFrameId(ee_name))
```
I'm gonna fix this in all scripts. 

I also agree that some index checking in robot wrapper would be appropriate. I'll try and add them.

andreadelprete, 2020-07-10:
I added the asserts in 96f0c8f95fc3ac1ee946d3ad581bb515d12b9678

closed 2020-07-10 by andreadelprete