stack-of-tasks/tsid #99: Installation issue

ddliugit, 2020-07-11:
I tried to install tsid on ubuntu 16.04 and used the `devel` branch. The error appears while I am using python bindings:

`>>> import tsid
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/tsid/__init__.py", line 1, in <module>
    from .libtsid_pywrap import *
ImportError: libeiquadprog.so: cannot open shared object file: No such file or directory
`

I installed the `eiquadprog` from the library under stack of tasks. 

Is the error caused by lacking of eiquadprog or wrong installation directory as I changed the install directory to `cmake .. -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr/local`?

olivier-stasse, 2020-07-11:
Dear @ddliugit, thanks for testing TSID.
We are lacking a bit of context.
From your description, it looks like you were able at some point to install eiquadprog on your machine in 
```
/usr/local
```
prefix.
If you can verifiy that ```/usr/local/lib/libeiquadprog.so```is present,
then to me it looks like you are missing 
```
/usr/local/lib
```
in your ```LD_LIBRARY_PATH``` environment variable.

ddliugit, 2020-07-11:
Yes, you are absolutely right. 

The problem is fixed after adding the path to the environment variable. However, the `tsid` cannot be imported since the following error:
`import tsid
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/tsid/__init__.py", line 1, in <module>
    from .libtsid_pywrap import *
ImportError: /usr/local/lib/python2.7/dist-packages/tsid/libtsid_pywrap.so: undefined symbol: _ZN7eigenpy9NumpyType4makeEP16tagPyArrayObjectb
`

andreadelprete, 2020-07-11:
Try to update eigenpy with
```
sudo apt install --reinstall robotpkg-eigenpy
```

ddliugit, 2020-07-11:
It doesn't work either.

```
jack@desktop:~$ sudo apt install --reinstall robotpkg-eigenpy
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  assimp-utils cython libavdevice-ffmpeg56 libcoin80v5 libcublas7.5
  libcudart7.5 libcufft7.5 libcufftw7.5 libcuinj64-7.5 libcurand7.5
  libcusolver7.5 libcusparse7.5 libfaad2 libgraphicsmagick-q16-3 libiso9660-8
  libmpcdec6 libmysofa0 libnppc7.5 libnppi7.5 libnpps7.5 libnvblas7.5
  libnvrtc7.5 libnvtoolsext1 libnvvm3 libopenscenegraph-dev
  libopenscenegraph100v5 libopenthreads-dev libopenthreads20 libthrust-dev
  libvcdinfo0 libvdpau-dev libx265-179 libx265-188 libxine2 libxine2-bin
  libxine2-doc libxine2-ffmpeg libxine2-misc-plugins libxine2-plugins
  nvidia-cuda-doc nvidia-cuda-gdb nvidia-opencl-dev nvidia-profiler
  nvidia-visual-profiler opencl-headers robotpkg-example-robot-data
  robotpkg-hpp-fcl+doc robotpkg-libccd robotpkg-octomap robotpkg-pinocchio
  robotpkg-py27-quadprog ros-kinetic-eigenpy
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 1 not upgraded.
Need to get 0 B/167 kB of archives.
After this operation, 0 B of additional disk space will be used.
(Reading database ... 415200 files and directories currently installed.)
Preparing to unpack .../robotpkg-eigenpy_1.4.0r2_amd64.deb ...
Unpacking robotpkg-eigenpy (1.4.0r2) over (1.4.0r2) ...
Setting up robotpkg-eigenpy (1.4.0r2) ...
jack@desktop:~$ python
Python 2.7.12 (default, Apr 15 2020, 17:07:12) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import tsid
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/tsid/__init__.py", line 1, in <module>
    from .libtsid_pywrap import *
ImportError: /usr/local/lib/python2.7/dist-packages/tsid/libtsid_pywrap.so: undefined symbol: _ZN7eigenpy9NumpyType4makeEP16tagPyArrayObjectb
```

andreadelprete, 2020-07-11:
Could you give us the output of this command? 
```
dpkg -l | grep robotpkg
```

ddliugit, 2020-07-11:
```
jack@desktop:~/workspace/dg_tutorial_with_turtlesim/src$ dpkg -l | grep robotpkgii  robotpkg-collada-dom                            2.4.0r1                                               amd64        C++ object representation of a COLLADA XML instance document
ii  robotpkg-eigenpy                                1.4.0r2                                               amd64        eigen binding to python
ii  robotpkg-example-robot-data                     3.4.2                                                 amd64        Robot URDFs for benchmarking and developed examples
ii  robotpkg-hpp-fcl+doc                            1.4.5                                                 amd64        Humanoid Path Planner (collision detection and distance computations)
ii  robotpkg-libccd                                 2.0                                                   amd64        collision detection
ii  robotpkg-octomap                                1.9.0                                                 amd64        An Efficient Probabilistic 3D Mapping Framework Based on Octrees
ii  robotpkg-openscenegraph                         3.6.5r3                                               amd64        open source high performance 3D graphics toolkit
ii  robotpkg-pinocchio                              2.4.6                                                 amd64        Efficient rigid body dynamics
ii  robotpkg-py27-qt4-gepetto-viewer                4.9.0r1                                               amd64        Humanoid Path Planner (Simple viewer based on OpenSceneGraph)
ii  robotpkg-py27-qt4-gepetto-viewer-corba          5.4.0r2                                               amd64        Humanoid Path Planner (Corba server for gepetto-viewer)
ii  robotpkg-py27-quadprog                          0.1.6r1                                               amd64        Solve a strictly convex quadratic program
ii  robotpkg-qt4-osgqt                              3.5.7r2                                               amd64        open source high performance 3D graphics toolkit

```

andreadelprete, 2020-07-11:
Try uninstalling `robotpkg-eigenpy` and installing instead `robotpkg-py27-eigenpy`

ddliugit, 2020-07-11:
It doesn't work as I used the `robotpkg-py27-eigenpy` at the beginning.
```
jack@desktop:~/repos/mpi_cmake_modules/build$ sudo apt-get remove robotpkg-eigenpy
[sudo] password for jack: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  assimp-utils cython libavdevice-ffmpeg56 libcoin80v5 libcublas7.5
  libcudart7.5 libcufft7.5 libcufftw7.5 libcuinj64-7.5 libcurand7.5
  libcusolver7.5 libcusparse7.5 libfaad2 libgraphicsmagick-q16-3 libiso9660-8
  libmpcdec6 libmysofa0 libnppc7.5 libnppi7.5 libnpps7.5 libnvblas7.5
  libnvrtc7.5 libnvtoolsext1 libnvvm3 libopenscenegraph-dev
  libopenscenegraph100v5 libopenthreads-dev libopenthreads20 libthrust-dev
  libvcdinfo0 libvdpau-dev libx265-179 libx265-188 libxine2 libxine2-bin
  libxine2-doc libxine2-ffmpeg libxine2-misc-plugins libxine2-plugins
  nvidia-cuda-doc nvidia-cuda-gdb nvidia-opencl-dev nvidia-profiler
  nvidia-visual-profiler opencl-headers robotpkg-example-robot-data
  robotpkg-hpp-fcl+doc robotpkg-libccd robotpkg-octomap robotpkg-pinocchio
  robotpkg-py27-quadprog ros-kinetic-eigenpy
Use 'sudo apt autoremove' to remove them.
The following packages will be REMOVED:
  robotpkg-eigenpy
0 upgraded, 0 newly installed, 1 to remove and 1 not upgraded.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
(Reading database ... 415199 files and directories currently installed.)
Removing robotpkg-eigenpy (1.4.0r2) ...
jack@desktop:~/repos/mpi_cmake_modules/build$ sudo apt-get install robotpkg-py27-eigenpy
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  assimp-utils cython libavdevice-ffmpeg56 libcoin80v5 libcublas7.5
  libcudart7.5 libcufft7.5 libcufftw7.5 libcuinj64-7.5 libcurand7.5
  libcusolver7.5 libcusparse7.5 libfaad2 libgraphicsmagick-q16-3 libiso9660-8
  libmpcdec6 libmysofa0 libnppc7.5 libnppi7.5 libnpps7.5 libnvblas7.5
  libnvrtc7.5 libnvtoolsext1 libnvvm3 libopenscenegraph-dev
  libopenscenegraph100v5 libopenthreads-dev libopenthreads20 libthrust-dev
  libvcdinfo0 libvdpau-dev libx265-179 libx265-188 libxine2 libxine2-bin
  libxine2-doc libxine2-ffmpeg libxine2-misc-plugins libxine2-plugins
  nvidia-cuda-doc nvidia-cuda-gdb nvidia-opencl-dev nvidia-profiler
  nvidia-visual-profiler opencl-headers robotpkg-example-robot-data
  robotpkg-hpp-fcl+doc robotpkg-libccd robotpkg-octomap robotpkg-pinocchio
  robotpkg-py27-quadprog ros-kinetic-eigenpy
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  robotpkg-py27-eigenpy
0 upgraded, 1 newly installed, 0 to remove and 1 not upgraded.
Need to get 911 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 http://robotpkg.openrobots.org/packages/debian/pub xenial/robotpkg amd64 robotpkg-py27-eigenpy amd64 2.4.0 [911 kB]
Fetched 911 kB in 2s (432 kB/s)                          
Selecting previously unselected package robotpkg-py27-eigenpy.
(Reading database ... 415140 files and directories currently installed.)
Preparing to unpack .../robotpkg-py27-eigenpy_2.4.0_amd64.deb ...
Unpacking robotpkg-py27-eigenpy (2.4.0) ...
Setting up robotpkg-py27-eigenpy (2.4.0) ...
jack@desktop:~/repos/mpi_cmake_modules/build$ python
Python 2.7.12 (default, Apr 15 2020, 17:07:12) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import tsid
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/tsid/__init__.py", line 1, in <module>
    from .libtsid_pywrap import *
ImportError: /usr/local/lib/python2.7/dist-packages/tsid/libtsid_pywrap.so: undefined symbol: _ZN7eigenpy9NumpyType4makeEP16tagPyArrayObjectb

```

andreadelprete, 2020-07-11:
The problem is that you have a too old version of eigenpy. These are the versions of the packages on my machine (same OS as you):
```
student@pinocchiovbox:~/devel/src/tsid/_build-RELEASE$ dpkg -l | grep robotpkg
ii  robotpkg-collada-dom                        2.3.1r2                                         amd64        C++ object representation of a COLLADA XML instance document
ii  robotpkg-eiquadprog                         1.2.0r1                                         amd64        algorithm of Goldfarb and Idnani for the solution of a QP
ii  robotpkg-hpp-fcl                            1.4.4                                           amd64        Humanoid Path Planner (collision detection and distance computations)
ii  robotpkg-libccd                             2.0                                             amd64        collision detection
ii  robotpkg-octomap                            1.9.0                                           amd64        An Efficient Probabilistic 3D Mapping Framework Based on Octrees
ii  robotpkg-omniorb                            4.2.4                                           amd64        ORB that implements the 2.6 specification of the CORBA
ii  robotpkg-osg-dae                            3.2.1r2                                         amd64        DAE module for OSG
ii  robotpkg-py35-catkin-pkg                    0.4.8                                           amd64        ROS standalone library for the catkin package
ii  robotpkg-py35-eigenpy                       2.3.0                                           amd64        Eigen bindings to python
ii  robotpkg-py35-omniorbpy                     4.2.4                                           amd64        Robust high-performance CORBA ORB for Python
ii  robotpkg-py35-qt4-gepetto-viewer            4.9.0                                           amd64        Humanoid Path Planner (Simple viewer based on OpenSceneGraph)
ii  robotpkg-py35-qt4-gepetto-viewer-corba      5.4.0                                           amd64        Humanoid Path Planner (Corba server for gepetto-viewer)
ii  robotpkg-py35-quadprog                      0.1.6                                           amd64        Solve a strictly convex quadratic program
ii  robotpkg-talos-data                         0.0.21r2                                        amd64        This packages provides data of the humanoid robot Pyrene in a friendly way for the SoT.
ii  robotpkg-ur5-description                    1.0.1                                           amd64        Rigid body model of a simple robot arm.
```
Try executing `sudo apt update`, then reinstall `robotpkg-py27-eigenpy`, then recompile TSID from a clean build directory.

ddliugit, 2020-07-11:
Yes, I updated, reinstalled and recompiled. The error still occurred. I am using python 2.7 right now, would that cause any problem?
I think you are using python 3.5

ddliugit, 2020-07-11:
My compiling procedure following `pinocchio`
```
cmake .. -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr/local
make -j4
make install
```
and then add the following command to the `~/.bashrc`
```
export PATH=/usr/local/bin:$PATH
export PKG_CONFIG_PATH =/usr/local/lib/pkgconfig:$PKG_CONFIG_PATH
export LD_LIBRARY_PATH=/usr/local/lib:$LD_LIBRARY_PATH
export PYTHONPATH =/usr/local/lib/python2.7/site-packages:$PYTHONPATH
```

andreadelprete, 2020-07-11:
Can you execute again `dpkg -l | grep robotpkg` and tell me the result? Because the problem seemed to be the too old version of eigenpy, so we should know whether that has been fixed by the update.

ddliugit, 2020-07-11:
```
jack@desktop:~/repos/robot_estimation$ dpkg -l | grep robotpkg
ii  robotpkg-collada-dom                            2.4.0r1                                               amd64        C++ object representation of a COLLADA XML instance document
ii  robotpkg-example-robot-data                     3.4.2                                                 amd64        Robot URDFs for benchmarking and developed examples
ii  robotpkg-hpp-fcl+doc                            1.4.5                                                 amd64        Humanoid Path Planner (collision detection and distance computations)
ii  robotpkg-libccd                                 2.0                                                   amd64        collision detection
ii  robotpkg-octomap                                1.9.0                                                 amd64        An Efficient Probabilistic 3D Mapping Framework Based on Octrees
ii  robotpkg-openscenegraph                         3.6.5r3                                               amd64        open source high performance 3D graphics toolkit
ii  robotpkg-pinocchio                              2.4.6                                                 amd64        Efficient rigid body dynamics
ii  robotpkg-py27-eigenpy                           2.4.0                                                 amd64        Eigen bindings to python
ii  robotpkg-py27-qt4-gepetto-viewer                4.9.0r1                                               amd64        Humanoid Path Planner (Simple viewer based on OpenSceneGraph)
ii  robotpkg-py27-qt4-gepetto-viewer-corba          5.4.0r2                                               amd64        Humanoid Path Planner (Corba server for gepetto-viewer)
ii  robotpkg-py27-quadprog                          0.1.6r1                                               amd64        Solve a strictly convex quadratic program
ii  robotpkg-qt4-osgqt                              3.5.7r2                                               amd64        open source high performance 3D graphics toolkit

```

ddliugit, 2020-07-12:
The problem is solved!

Thanks for your help!

closed 2020-07-12 by ddliugit