stack-of-tasks/tsid #103: Delete solvers on mac

jbmouret, 2020-07-22:
Hi

I am trying to delete the solver created from the factory. On Mac, I get an "illegal instruction".

To reproduce, add:
```c++
delete solver;
```
at the end of `./tests/hqp_solvers`.

I get:
```
./tests/hqp_solvers         
Running 1 test case...
test_eiquadprog_classic_vs_rt_vs_fast
Gonna perform 1000 tests with 60 variables, 36 equalities, 40 inequalities
unknown location:0: fatal error: in "hqp_solversTest/test_eiquadprog_classic_vs_rt_vs_fast": signal: illegal opcode; address of failing instruction: 0x1054d1ed2
/Users/jmouret/git/laas/tsid/tests/hqp_solvers.cpp:305: last checkpoint

*** 1 failure is detected in the test module "hqp_solversTest"
```

It seems to work fine on our Linux (Ubuntu). I tried to compile with `-fno-stack-check` but it did not help so far.

Details:
```
clang --version
Apple clang version 11.0.3 (clang-1103.0.32.62)
Target: x86_64-apple-darwin19.6.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin

MacOS 10.15.6 (last version in Jul. 2020).

```

I had issues in this test too when I tried to compile with `-march=native` (to enable eigen's vectorization), so I guess this is some kind of alignment issue.

olivier-stasse, 2020-07-22:
Trying to recompile tsid on Mac and launch the test with
```bash
colcon build --packages-select tsid --cmake-args -DPYTHON_EXECUTABLE=/usr/local/opt/python@3.8/bin/python3
colcon test --packages-select tsid
```
gave
```bash
Assertion failed: (model.check(data) && "data is not consistent with model."), function computeAllTerms, file /usr/local/include/pinocchio/algorithm/compute-all-terms.hpp, line 209.
```
So I will try to see if I can reproduce your bug.

olivier-stasse, 2020-07-22:
The previous test did not specify the optimization flag so I retried with 
```bash
colcon build --packages-select tsid --cmake-args -DPYTHON_EXECUTABLE=/usr/local/opt/python@3.8/bin/python3 -DCMAKE_BUILD_TYPE=RELEASE
```
the compiling options are:
```bash
 /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++  -DBOOST_MPL_LIMIT_LIST_SIZE=30 -DBOOST_MPL_LIMIT_VECTOR_SIZE=30 -DEIGEN_RUNTIME_NO_MALLOC -DPINOCCHIO_WITH_HPP_FCL -DPINOCCHIO_WITH_URDFDOM -Dtsid_EXPORTS -I/Users/ostasse/sot_ws/build/tsid -I/Users/ostasse/sot_ws/build/tsid/include -I/Users/ostasse/sot_ws/src/sot/tsid/include -isystem /usr/local/Cellar/eigen/3.3.7/include/eigen3 -isystem /usr/local/include -isystem /usr/local/Cellar/console_bridge/1.0.1/include -isystem /usr/local/Cellar/urdfdom_headers/1.0.5/include -isystem /usr/local/Cellar/urdfdom/1.0.4_2/include -isystem /usr/local/Cellar/octomap/1.9.5/include -isystem /usr/local/Cellar/hpp-fcl/1.4.5/include -isystem /usr/local/include/eigen3 -isystem /Users/ostasse/sot_ws/install/eiquadprog/include  -pedantic -Wno-long-long -Wall -Wextra -Wcast-align -Wcast-qual -Wformat -Wwrite-strings -Wconversion  -O3 -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk -fPIC   -DHPP_FCL_HAVE_OCTOMAP -DFCL_HAVE_OCTOMAP -DOCTOMAP_MAJOR_VERSION=1 -DOCTOMAP_MINOR_VERSION=9 -DOCTOMAP_PATCH_VERSION=5 -o CMakeFiles/tsid.dir/src/solvers/solver-HQP-base.cpp.o -c /Users/ostasse/sot_ws/src/sot/tsid/src/solvers/solver-HQP-base.cpp
```
The test passed.
The clang output is the same than yours:
```bash
Apple clang version 11.0.3 (clang-1103.0.32.62)
Target: x86_64-apple-darwin19.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin

Mac OS X 10.15.6
```

jbmouret, 2020-07-22:
Thanks.

```
cmake ../ -DCMAKE_INSTALL_PREFIX=$HOME -DCMAKE_BUILD_TYPE=RELEASE -DBUILD_PYTHON_INTERFACE=OFF
```

```
cd /Users/jmouret/git/laas/tsid/build/tests && /Library/Developer/CommandLineTools/usr/bin/c++  -DBOOST_MPL_LIMIT_LIST_SIZE=30 -DBOOST_MPL_LIMIT_VECTOR_SIZE=30 -DEIGEN_RUNTIME_NO_MALLOC -DPINOCCHIO_WITH_URDFDOM -I/Users/jmouret/git/laas/tsid/build -I/Users/jmouret/git/laas/tsid/build/include -I/Users/jmouret/git/laas/tsid/include -isystem /usr/local/include -isystem /usr/local/include/eigen3 -isystem /Users/jmouret/include -isystem /usr/local/Cellar/urdfdom/1.0.4_2/include -isystem /usr/local/lib/urdfdom_headers/cmake/../../../include  -pedantic -Wno-long-long -Wall -Wextra -Wcast-align -Wcast-qual -Wformat -Wwrite-strings -Wconversion  -O3 -DNDEBUG -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk    -DBOOST_TEST_DYN_LINK -DBOOST_TEST_MODULE=hqp_solversTest -o CMakeFiles/hqp_solvers.dir/hqp_solvers.cpp.o -c /Users/jmouret/git/laas/tsid/tests/hqp_solvers.cpp
```

I think the compilation is the same. Did you modify the test to add the `delete`? the current test works, the delete is what breaks things (whereas I think it should not).

If the delete works for you, this is probably some installation problem on my side (maybe some issues with my installation of Pinocchio or eiquadprog)..

olivier-stasse, 2020-07-22:
My bad,  I did not add the delete. I will retry the last test with delete.

olivier-stasse, 2020-07-22:
I confirm that I have the same error.

olivier-stasse, 2020-07-22:
Could you tell on which branch you are ?
I would guess master.

jbmouret, 2020-07-22:
Yes, I did not try any other branch.

olivier-stasse, 2020-07-22:
It comes from the fact that solver-HQP-base.hpp does not have a virtual destructor while having virtual methods.

If you use the version from the fork here:
https://github.com/olivier-stasse/tsid/tree/devel
could you confirm that it is fixing the problem on your side ?

I did a pull request to fix the matter. I will remove the draft tag as soon as it works for you.

jbmouret, 2020-07-22:
Thanks! A classic (I think we should get a warning for this, but I did not see it...).

Yes, it works for me.

olivier-stasse, 2020-07-22:
> Thanks! A classic (I think we should get a warning for this, but I did not see it...).
Totally agree, I am surprised such mistake was still here.
 
> Yes, it works for me.
Thanks then I will remove the draft tag from the PR.

I will close the issue.
Feel free to reopen it if needed.

PR #104 (2020-07-23): [test] Delete solvers
Fix solver-HQP-base.hpp without virtual destructor.
Thanks to JB Mouret who reported the bug.
Fix #103
closed 2020-07-22 by olivier-stasse