stack-of-tasks/tsid #110: Memory leaks?

jbmouret, 2020-09-23:
Hi

It seems that TSID is leaking quite a lot of memory. Valgrind says that for the tests/tsid-formulation:

```
==59085== LEAK SUMMARY:
==59085==    definitely lost: 229,104 bytes in 17 blocks
==59085==    indirectly lost: 2,912,485 bytes in 1,897 blocks
==59085==      possibly lost: 6,816 bytes in 1 blocks
==59085==    still reachable: 72,704 bytes in 1 blocks
==59085==         suppressed: 0 bytes in 0 blocks
==59085== Reachable blocks (those to which a pointer was found) are not shown.
==59085== To see them, rerun with: --leak-check=full --show-leak-kinds=all
```
=> so, about 2Mb for the test (it might look like nothing, but our code create TSID objects about 50-100k times, which means about 100Gb can be lost!

See the full report below. From a quick look, it seems that the tsid formulation calls many new, but no delete (see https://github.com/stack-of-tasks/tsid/blob/master/src/formulations/inverse-dynamics-formulation-acc-force.cpp).

=> is it "normal"? It would not be hard to call delete in the destructor or put everything in shared_ptr, I guess.

```
==59754== Memcheck, a memory error detector
==59754== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==59754== Using Valgrind-3.16.1 and LibVEX; rerun with -h for copyright info
==59754== Command: tests/tsid-formulation
==59754== 
Running 4 test cases...

*** test_invdyn_formulation_acc_force_remove_contact ***
q: 0 0 0 0 0 0 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
[SolverHQuadProg.solver-eiquadprog] Resizing equality constraints from 0 to 18
[SolverHQuadProg.solver-eiquadprog] Resizing inequality constraints from 0 to 71
[SolverHQuadProg.solver-eiquadprog] Resizing Hessian from 0 to 61
Level 0
 - base-dynamics: w=1, equality, 6x61
 - contact_rfoot_motion_task: w=1, equality, 6x61
 - contact_rfoot_force_constraint: w=1, inequality, 17x61
 - contact_lfoot_motion_task: w=1, equality, 6x61
 - contact_lfoot_force_constraint: w=1, inequality, 17x61
 - task-joint-bounds: w=1, inequality, 37x61
Level 1
 - contact_rfoot_force_reg_task: w=1e-05, equality, 6x61
 - contact_lfoot_force_reg_task: w=1e-05, equality, 6x61
 - task-com: w=1, equality, 3x61
 - task-posture: w=0.01, equality, 31x61
 - task-right-foot: w=1000, equality, 6x61

Time 0
  contact_rfoot force: 392.616 	  contact_lfoot force: 5 	task-com err: 0.1 	v=0	 dv=10.1726

### TEST FINISHED ###
v(37x1): [ 0.0006,     0.2,  -0.004,   2e-06, -0.0004,    0.03,   -0.03,    -0.3,  -0.003,   3e-05,   0.003,     0.3,   -0.03,    -0.3,   0.005, -0.0002,  -0.005,     0.3,    0.05,  -0.006,    0.02,   5e-05,   0.006,   2e-05,  0.0007,  -9e-05,   0.001,  -1e-05, -0.0001, -0.0006,   0.002,    0.02,   3e-05,   0.006,   3e-05,  0.0007,  -7e-05]
Final   CoM position:  0.0219732 0.00825469   0.665849
Desired CoM position: 0.0219541       0.1  0.665915

*** test_invdyn_formulation_acc_force ***
q: 0 0 0 0 0 0 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
[SolverHQuadProg.solver-eiquadprog] Resizing equality constraints from 0 to 18
[SolverHQuadProg.solver-eiquadprog] Resizing inequality constraints from 0 to 71
[SolverHQuadProg.solver-eiquadprog] Resizing Hessian from 0 to 61
nVar 61
nEq 18
nIn 71
Initial CoM position:    0.0219541 -3.46945e-18     0.665915
Initial RF position:   R =
1 0 0
0 1 0
0 0 1
  p =       0  -0.096 0.02996

[... 395 lines cut ...]
==59754==    by 0x78D4D1: tsid_formulationTest::test_invdyn_formulation_acc_force_remove_contact::test_method() (tsid-formulation.cpp:208)
==59754==    by 0x78D3FF: tsid_formulationTest::test_invdyn_formulation_acc_force_remove_contact_invoker() (tsid-formulation.cpp:197)
==59754==    by 0x83DE4D: boost::unit_test::ut_detail::unused boost::unit_test::ut_detail::invoker<boost::unit_test::ut_detail::unused>::invoke<void (*)()>(void (*&)()) (callback.hpp:56)
==59754==    by 0x83B0E4: boost::unit_test::ut_detail::callback0_impl_t<boost::unit_test::ut_detail::unused, void (*)()>::invoke() (callback.hpp:89)
==59754==    by 0x6356CB0: operator() (callback.hpp:118)
==59754==    by 0x6356CB0: operator() (unit_test_monitor.ipp:41)
==59754==    by 0x6356CB0: invoke<boost::unit_test::(anonymous namespace)::zero_return_wrapper_t<boost::unit_test::callback0<> > > (callback.hpp:42)
==59754==    by 0x6356CB0: boost::unit_test::ut_detail::callback0_impl_t<int, boost::unit_test::(anonymous namespace)::zero_return_wrapper_t<boost::unit_test::callback0<boost::unit_test::ut_detail::unused> > >::invoke() (callback.hpp:89)
==59754==    by 0x6336995: operator() (callback.hpp:118)
==59754==    by 0x6336995: do_invoke<boost::scoped_ptr<boost::detail::translate_exception_base>, boost::unit_test::callback0<int> > (execution_monitor.ipp:281)
==59754==    by 0x6336995: boost::execution_monitor::catch_signals(boost::unit_test::callback0<int> const&) (execution_monitor.ipp:885)
==59754==    by 0x63371B2: boost::execution_monitor::execute(boost::unit_test::callback0<int> const&) (execution_monitor.ipp:1211)
==59754== 
==59754== LEAK SUMMARY:
==59754==    definitely lost: 229,104 bytes in 17 blocks
==59754==    indirectly lost: 2,919,005 bytes in 1,897 blocks
==59754==      possibly lost: 296 bytes in 1 blocks
==59754==    still reachable: 72,704 bytes in 1 blocks
==59754==         suppressed: 0 bytes in 0 blocks
==59754== Reachable blocks (those to which a pointer was found) are not shown.
==59754== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==59754== 
==59754== For lists of detected and suppressed errors, rerun with: -s
==59754== ERROR SUMMARY: 18 errors from 18 contexts (suppressed: 0 from 0)
```

nim65s, 2020-09-23:
Hi @jbmouret, and thanks for this report and investigation.

I totally agree, we should use more shared_ptr.

I could handle this task, but not soon, so if anyone wants to help a bit on that, this would be really welcome :)

jbmouret, 2020-09-23:
OK, I will put it in my (long) TODO list.

andreadelprete, 2020-09-23:
Agreed. When I first implemented TSID I didn't have in mind this use-case, so I didn't take care of deallocating memory. My bad. Unfortunately now that the semester started I'm also very busy, so I don't know when I could find the time to do it. Worst case could be January for me.

jbmouret, 2020-09-24:
@andreadelprete OK. At least I know that there is no secret deallocator :) I will try to work on this and submit a PL.

jbmouret, 2020-09-24:
Is there a reason for not compiling in c++11 ? (14 or 17 would be even better). We do not have shared_ptr in std:: in c++98, except if we use boost.

jcarpent, 2020-09-24:
Since of URDFDOM >= 1.0.0, we have to rely on C++11 at the Pinocchio level. So, except if @nim65s or @olivier-stasse or @andreadelprete do agree, we may go toward std::shared_ptr and C++11. But boost::shared_ptr remains a good candiate I would say.

olivier-stasse, 2020-09-24:
Until this month, our TALOS was in 16.04 LTS. And by default 16.04 LTS is in C++98. We had several problems with some boost libraries such as boost::variant not behaving nicely when mixing C++98 and C++11. But all of our robots are now at least on 18.04 LTS. So I am fine with C++11.

jbmouret, 2020-09-24:
I've moved all the pointers to shared_ptr and there is no leak anymore! (and yes, this means c++11)

```
=56088== LEAK SUMMARY:
==56088==    definitely lost: 0 bytes in 0 blocks
==56088==    indirectly lost: 0 bytes in 0 blocks
==56088==      possibly lost: 0 bytes in 0 blocks
```

https://github.com/resibots/tsid/tree/fix_leaks 

I still need to have a look at the python interface, as we usually do not attempt to compile it.

olivier-stasse, 2020-09-25:
It looks nice. Thanks.

andreadelprete, 2020-09-25:
I am fine with the switch to c++11

nim65s, 2020-09-25:
closed by #112

PR #112 (2020-09-25): Fix memory leaks by using std::shared_ptr<>
This PL uses `std::shared_ptr<>` instead of classic pointers (which were never deleted). This is mostly in `inverse-dynamics-formulation-base.cpp`.
This should fix #110.

I think the "public" API should not have changed, but please double-check (our code worked out of the box). The python side seems to work (but I did not test in details).

We could consider a few API changes. For instance, the solver factory could return a shared_ptr instead of a pointer. There are also a few

We have to move to c++11 for std::shared_ptr (I also used a few auto, sin [... cut]
closed 2020-09-25 by nim65s