stack-of-tasks/tsid #115: CoP and DCM tasks

NahuelVilla, 2020-10-22:
Hi, I am using tsid on its python version. In my implementation I need the robot to track certain desired Divergent Component of the Motion (DCM) and Center of Pressure (CoP) on a given contact surface.

Currently, some work around could be done for both tasks, such as implementing a force distributor that represents the CoP as four forces on vertices of a rectangular contact surface to use a Contact6d. But I would like to have the DCMTask and CoPTask which seems to be available in the C++ version of tsid.

So, I would like to know your recommendations. Is the work around a fast and efficient solution in python? Does someone have the python bindings for these tasks? What is the easiest form to track DCM and CoP in the python tsid?

thanks.

andreadelprete, 2020-10-22:
Hi @NahuelVilla, as far as I know there're no DCM or CoP tasks implemented in C++. Could you give me a pointer to them?
Anyway, if they're already implemented in C++, implementing the Python bindings is rather simple and fast. If instead you also need to take care of the C++ implementation, that's a bit more work. In any case, if you're willing to do it, I can guide you. Otherwise I can also do it, but it may take me some time.

NahuelVilla, 2020-10-22:
Hi Andreas, thanks for answering so fast.

I meant this task for the DCM:
https://github.com/stack-of-tasks/tsid/blob/master/src/tasks/task-capture-point-inequality.cpp
and for the CoP I wouldn't know. Anyway, I am looking rather for an equality constraint task similar to TaskComEquality.

I am sticking to python code, developing C++ for me now would be much more time.

andreadelprete, 2020-10-23:
I think that transforming the TaskComEquality into a TaskDcmEquality should be rather simple. However, if you don't wanna write C++, an easy workaround is to compute the desired CoM acceleration in python (based on the DCM feedback law), and then use a TaskComEquality, imposing the computed acceleration as feedforward, and setting the task feedback gains to zero.
For the CoP the this is a bit trickier if you have multiple contacts, but you can do something similar.

nmansard, 2020-11-24:
Ok, thanks for the info. So in a first time, we will implement a force distributor to decide what should be the contact forces to obtain the desired COP, and put a task in TSID to track the forces computed by the distributor. Does that seem realistic to you?

In a second time, we might be able to implement a COP task in TSID. Would that make sense?

andreadelprete, 2020-11-24:
Yes, that seems a good plan to me.

nmansard, 2020-11-24:
Would you put a cost on the 3d forces at each foot corner? Or a cost on the 6d force directly? Could you point us toward the name of the task to use in the second case, if it exists?

andreadelprete, 2020-11-24:
The force regularization task is already embedded in any rigid contact. If it's for a biped, I imagine you wanna use [contact6d](https://github.com/stack-of-tasks/tsid/blob/master/src/contacts/contact-6d.cpp). The reference force would be a 6d wrench. You can set it using [this method](https://github.com/stack-of-tasks/tsid/blob/master/src/contacts/contact-6d.cpp#L216).

An alternative you might wanna consider is to combine an angular momentum task with a CoM task. These two together would allow you to indirectly control the CoP, without pre-specifying the force distribution.

nmansard, 2020-11-24:
yes, I agree with you about combining tasks. We could do that, but I am afraid of the side effects, in particular when debugging. We will try first with the distribution, and, if you are ok, come back to you later to possibly add a cop tasks. 

I think we can close the issue now. Thanks a lot for your help.

closed 2020-11-24 by nmansard

andreadelprete, 2020-11-24:
Ok, after the semester is finished, in January, I should have time to implement the CoP task!

andreadelprete, 2020-11-25:
Beware there was a small bug in Contact6d. I've just fixed it with the last commit, so you may wanna use the latest version of the code (devel branch). This bug only affects cases in which the reference contact forces are not zero, which I think is gonna be your case.