stack-of-tasks/tsid #118: mask of task-se3-equality

dalinel, 2020-12-08:
Hello,
I have a problem about the task-se3-equality

I am controlling a gripper position and orientation of a robot with a full se3 reference and it works
But  if I put a x and y mask and then pass highly abnormal references for the x and y (ie -10000 and 800) :
The solver output is weird or doesn't find a solution

With a mask on the x and y, I would have expected the task to ignore the false references and proceed as if I had not given those

I think that there is some variable coupling somewhere which makes the mask at the end inefficient for high abnormal references. Maybe with  ``` errorInSE3(oMi, m_M_ref, m_p_error) ```  ?

I also have tried to apply the following mask before the ``` errorInSE3(oMi, m_M_ref, m_p_error); ``` line :
```
  for (int i=0; i<3; i++)
  {
    if (m_mask[i] == 0.)
      m_M_ref.translation()(i) = oMi.translation()(i);
  }

  double roll = m_M_ref.rotation().eulerAngles(0,1,2)(0);
  if (m_mask[3] == 0.)
    roll = oMi.rotation().eulerAngles(0,1,2)(0);
  double pitch = m_M_ref.rotation().eulerAngles(0,1,2)(1);
  if (m_mask[4] == 0.)
    pitch = oMi.rotation().eulerAngles(0,1,2)(1);
  double yaw = m_M_ref.rotation().eulerAngles(0,1,2)(2);
  if (m_mask[5] == 0.)
    yaw = oMi.rotation().eulerAngles(0,1,2)(2);

  Eigen::Quaterniond quat = Eigen::AngleAxisd(roll, Eigen::Vector3d::UnitX()) * Eigen::AngleAxisd(pitch, Eigen::Vector3d::UnitY()) * Eigen::AngleAxisd(yaw, Eigen::Vector3d::UnitZ());
  m_M_ref = pinocchio::SE3(quat, m_M_ref.translation());
```

When I do that with the abnormal x and y references (and mask), the solver finds a solution and the z, roll pitch and yaw error is next to zero

Do you have any hints or ideas about what is going on ?

andreadelprete, 2020-12-09:
My guess is that you're expressing the task in local frame, so also the mask is applied in local frame. As the local frame is probably not aligned with the world frame (which is the one used for expressing the reference) things don't work as you expect. Try to switch the task to the global frame by using the method [useLocalFrame](https://github.com/stack-of-tasks/tsid/blob/master/include/tsid/tasks/task-se3-equality.hpp#L108).

dalinel, 2020-12-09:
I have the problem with m_local_frame to true or false
If the value of x and y are good enough (even if wrong) the mask seems to work.
But when I put highly abnormal values it fails

andreadelprete, 2020-12-09:
Maybe there could be some numerical error due to the fact that the error is first computed in local frame (using `errorInSE3`) and then projected in world frame (using `m_wMl.toActionMatrix()`). Could you try to print the values of `m_p_error_vec`, `m_p_error` and `m_a_des` in [this line](https://github.com/stack-of-tasks/tsid/blob/master/src/tasks/task-se3-equality.cpp#L204)?
Having large values of the reference only in xy, you should get large values only in the first two entries of `m_p_error_vec` and `m_a_des`, but possibly in all the first 3 entries of `m_p_error`. Could you check that's the case?

dalinel, 2020-12-09:
At the same time step I have :

with correct reference (xref = 0.0779931, yref = -0.315164, zref = 0.92661)
```
frame 81: m_p_error_vec -5.61677e-06  2.36178e-05 -2.51817e-05  3.31814e-05 -0.000384383 -3.36401e-05
frame 81: m_a_des   0.105764 -0.0628411   0.170859  -0.101975  0.0721051   0.351831
frame 81: m_p_error 
  v = -2.36797e-06  2.36622e-05 -2.56507e-05
  w =  4.56884e-05 -0.000383419 -2.97537e-05
```

with highly abnormal reference on x and y  : (xref = -100000, yref = 800, zref = 0.92661)
```
frame 81: m_p_error_vec      99999.9     -802.363      18.3816  2.77719e-05 -0.000367955 -3.97583e-05
frame 81: m_a_des      -3e+08 2.40709e+06    -55144.5  -0.0858434   0.0226373    0.370224
frame 81: m_p_error 
  v = 98920.4 1196.29 14626.6
  w =  4.08942e-05 -0.000367063  -3.6541e-05
```

andreadelprete, 2020-12-09:
It should be fixed by the last commit. I'll come back later with an explanation of what was wrong with the code.

dalinel, 2020-12-09:
Thank you, I have tested and it works now

andreadelprete, 2020-12-09:
There were two problems with the code.

The first one was simply a bug. The position error was expressed in the reference frame, and when converting it to the world frame I was instead assuming it was expressed in the actual local frame.

The second one was not a bug, but was simply the fact of using the `log6` function for computing the error. This function indeed computes the error in a way that couples translation and orientation, so when the orientation error was not null, then the large errors of the xy components was also affecting the z component. To avoid this, I've switched to the `log3` function to compute the orientation error, whereas I use a simple difference for the translation error.

andreadelprete, 2020-12-11:
I've added this code after line 198:
```
cout<<"m_p_error_vec="<<m_p_error_vec.head<3>().transpose()<<endl;
cout<<"oMi-m_M_ref  ="<<(oMi.translation()-m_M_ref.translation()).transpose()<<endl;
```
and I always get the same values:
```
m_p_error_vec=  -0.0999804 -1.12257e-07 -1.63321e-07
oMi-m_M_ref  =  -0.0999804 -1.12257e-07 -1.63321e-07
m_p_error_vec=  -0.0999223 -4.40853e-07 -6.41018e-07
oMi-m_M_ref  =  -0.0999223 -4.40853e-07 -6.41018e-07
m_p_error_vec=  -0.0998272 -9.69816e-07 -1.40918e-06
oMi-m_M_ref  =  -0.0998272 -9.69816e-07 -1.40918e-06
m_p_error_vec=  -0.0996967 -1.68393e-06 -2.44511e-06
oMi-m_M_ref  =  -0.0996967 -1.68393e-06 -2.44511e-06
m_p_error_vec=  -0.0995322  -2.5687e-06 -3.72727e-06
oMi-m_M_ref  =  -0.0995322  -2.5687e-06 -3.72727e-06
m_p_error_vec=   -0.099335 -3.61033e-06 -5.23522e-06
oMi-m_M_ref  =   -0.099335 -3.61033e-06 -5.23522e-06
m_p_error_vec=  -0.0991066  -4.7957e-06 -6.94958e-06
oMi-m_M_ref  =  -0.0991066  -4.7957e-06 -6.94958e-06
m_p_error_vec=  -0.0988483 -6.11231e-06 -8.85199e-06
oMi-m_M_ref  =  -0.0988483 -6.11231e-06 -8.85199e-06
m_p_error_vec=  -0.0985612 -7.54832e-06  -1.0925e-05
oMi-m_M_ref  =  -0.0985612 -7.54832e-06  -1.0925e-05
m_p_error_vec=  -0.0982468 -9.09242e-06 -1.31522e-05
oMi-m_M_ref  =  -0.0982468 -9.09242e-06 -1.31522e-05
m_p_error_vec=  -0.0979061 -1.07339e-05 -1.55179e-05
oMi-m_M_ref  =  -0.0979061 -1.07339e-05 -1.55179e-05
m_p_error_vec=  -0.0975403 -1.24627e-05 -1.80073e-05
oMi-m_M_ref  =  -0.0975403 -1.24627e-05 -1.80073e-05
m_p_error_vec=  -0.0971506 -1.42691e-05 -2.06065e-05
oMi-m_M_ref  =  -0.0971506 -1.42691e-05 -2.06065e-05
m_p_error_vec=  -0.0967381 -1.61439e-05 -2.33023e-05
oMi-m_M_ref  =  -0.0967381 -1.61439e-05 -2.33023e-05
m_p_error_vec=  -0.0963037 -1.80786e-05 -2.60821e-05
oMi-m_M_ref  =  -0.0963037 -1.80786e-05 -2.60821e-05
m_p_error_vec=  -0.0958486 -2.00649e-05 -2.89342e-05
oMi-m_M_ref  =  -0.0958486 -2.00649e-05 -2.89342e-05
m_p_error_vec=  -0.0953737 -2.20951e-05 -3.18475e-05
oMi-m_M_ref  =  -0.0953737 -2.20951e-05 -3.18475e-05
m_p_error_vec=    -0.09488 -2.41619e-05 -3.48114e-05
oMi-m_M_ref  =    -0.09488 -2.41619e-05 -3.48114e-05
m_p_error_vec=  -0.0943684 -2.62584e-05 -3.78162e-05
oMi-m_M_ref  =  -0.0943684 -2.62584e-05 -3.78162e-05
m_p_error_vec=  -0.0938398  -2.8378e-05 -4.08525e-05
oMi-m_M_ref  =  -0.0938398  -2.8378e-05 -4.08525e-05
m_p_error_vec=  -0.0932951 -3.05147e-05 -4.39115e-05
oMi-m_M_ref  =  -0.0932951 -3.05147e-05 -4.39115e-05
m_p_error_vec=  -0.0927351 -3.26627e-05 -4.69852e-05
oMi-m_M_ref  =  -0.0927351 -3.26627e-05 -4.69852e-05
m_p_error_vec=  -0.0921607 -3.48165e-05 -5.00657e-05
oMi-m_M_ref  =  -0.0921607 -3.48165e-05 -5.00657e-05
m_p_error_vec=  -0.0915726  -3.6971e-05 -5.31458e-05
oMi-m_M_ref  =  -0.0915726  -3.6971e-05 -5.31458e-05
m_p_error_vec=  -0.0909717 -3.91214e-05 -5.62189e-05
oMi-m_M_ref  =  -0.0909717 -3.91214e-05 -5.62189e-05
m_p_error_vec=  -0.0903586 -4.12633e-05 -5.92785e-05
oMi-m_M_ref  =  -0.0903586 -4.12633e-05 -5.92785e-05
m_p_error_vec=  -0.0897341 -4.33925e-05 -6.23188e-05
oMi-m_M_ref  =  -0.0897341 -4.33925e-05 -6.23188e-05
m_p_error_vec=  -0.0890988 -4.55049e-05 -6.53342e-05
oMi-m_M_ref  =  -0.0890988 -4.55049e-05 -6.53342e-05
m_p_error_vec=  -0.0884534 -4.75971e-05 -6.83197e-05
oMi-m_M_ref  =  -0.0884534 -4.75971e-05 -6.83197e-05
m_p_error_vec=  -0.0877987 -4.96656e-05 -7.12706e-05
oMi-m_M_ref  =  -0.0877987 -4.96656e-05 -7.12706e-05
m_p_error_vec=  -0.0871351 -5.17072e-05 -7.41823e-05
oMi-m_M_ref  =  -0.0871351 -5.17072e-05 -7.41823e-05
m_p_error_vec=  -0.0864634  -5.3719e-05 -7.70508e-05
oMi-m_M_ref  =  -0.0864634  -5.3719e-05 -7.70508e-05
m_p_error_vec=   -0.085784 -5.56983e-05 -7.98724e-05
oMi-m_M_ref  =   -0.085784 -5.56983e-05 -7.98724e-05
m_p_error_vec=  -0.0850976 -5.76426e-05 -8.26435e-05
oMi-m_M_ref  =  -0.0850976 -5.76426e-05 -8.26435e-05
m_p_error_vec=  -0.0844047 -5.95496e-05 -8.53611e-05
oMi-m_M_ref  =  -0.0844047 -5.95496e-05 -8.53611e-05
m_p_error_vec=  -0.0837058 -6.14172e-05 -8.80221e-05
oMi-m_M_ref  =  -0.0837058 -6.14172e-05 -8.80221e-05
m_p_error_vec=  -0.0830015 -6.32435e-05  -9.0624e-05
oMi-m_M_ref  =  -0.0830015 -6.32435e-05  -9.0624e-05
m_p_error_vec=  -0.0822922 -6.50267e-05 -9.31643e-05
oMi-m_M_ref  =  -0.0822922 -6.50267e-05 -9.31643e-05
m_p_error_vec=  -0.0815784 -6.67652e-05 -9.56409e-05
oMi-m_M_ref  =  -0.0815784 -6.67652e-05 -9.56409e-05
m_p_error_vec=  -0.0808605 -6.84577e-05 -9.80517e-05
oMi-m_M_ref  =  -0.0808605 -6.84577e-05 -9.80517e-05
m_p_error_vec=  -0.0801391 -7.01028e-05 -0.000100395
oMi-m_M_ref  =  -0.0801391 -7.01028e-05 -0.000100395
m_p_error_vec=  -0.0794144 -7.16994e-05  -0.00010267
oMi-m_M_ref  =  -0.0794144 -7.16994e-05  -0.00010267
m_p_error_vec=   -0.078687 -7.32465e-05 -0.000104874
oMi-m_M_ref  =   -0.078687 -7.32465e-05 -0.000104874
m_p_error_vec=  -0.0779572 -7.47432e-05 -0.000107006
oMi-m_M_ref  =  -0.0779572 -7.47432e-05 -0.000107006
m_p_error_vec=  -0.0772254 -7.61888e-05 -0.000109066
oMi-m_M_ref  =  -0.0772254 -7.61888e-05 -0.000109066
m_p_error_vec=   -0.076492 -7.75826e-05 -0.000111053
oMi-m_M_ref  =   -0.076492 -7.75826e-05 -0.000111053
m_p_error_vec=  -0.0757572 -7.89242e-05 -0.000112965
oMi-m_M_ref  =  -0.0757572 -7.89242e-05 -0.000112965
m_p_error_vec=  -0.0750216  -8.0213e-05 -0.000114803
oMi-m_M_ref  =  -0.0750216  -8.0213e-05 -0.000114803
m_p_error_vec=  -0.0742853 -8.14488e-05 -0.000116565
oMi-m_M_ref  =  -0.0742853 -8.14488e-05 -0.000116565
```
Can you help me reproduce the problem?

dalinel, 2020-12-11:
At the same time step I have
with a correct reference : 
```
right_hand_reference
  R =
   0.999732   0.0231463 1.82246e-06
 -0.0231415    0.999533   0.0200125
0.000461517  -0.0200072      0.9998
  p = 0.0534311 -0.204592  0.914049

frame 81 m_p_error_vec=  0.118956  -0.079308 -0.0123425
frame 81 oMi-m_M_ref  =  0.118956 -0.0793079 -0.0123425
```

with a wrong reference:
```
right_hand_reference
  R =
   0.999732   0.0231463 1.82246e-06
 -0.0231415    0.999533   0.0200125
0.000461517  -0.0200072      0.9998
  p =    10000     -800 0.914049

frame 81 m_p_error_vec=  -9999.83    799.715 -0.0135848
frame 81 oMi-m_M_ref  =  -9999.83    799.716 -0.0123425
```

with a very wrong reference:
```
right_hand_reference
  R =
   0.999732   0.0231463 1.82246e-06
 -0.0231415    0.999533   0.0200125
0.000461517  -0.0200072      0.9998
  p =  8.09836e+18 -1.01865e+19     0.914049

frame 81 m_p_error_vec=-8.09836e+18  1.01865e+19 -1.15365e+12
frame 81 oMi-m_M_ref  =-8.09836e+18  1.01865e+19   -0.0123425
```

andreadelprete, 2020-12-11:
I see. I've carried out some tests, and it seems that these errors are simply numerical, and indeed they only become significant if you input veeeery large values. Even with a reference of 1e10 I get very small errors, which do not affect the robot behavior:
```
m_p_error_vec=5.85318e-05       1e+10 2.14577e-06
oMi-m_M_ref  =5.81238e-05       1e+10 2.45136e-06
```
Of course I could change the code such that this specific numerical error doesn't occur anymore, but there're always gonna be countless lines in the code where numerical errors would occur if you start using crazy large numbers. This is rather philosophical, but I don't think it's important that TSID gives perfect results for crazy references, unless of course there's a good reason for using crazy references in robotics that I'm not seeing. So before trying to avoid this numerical issue I'd like to know the motivation behind this.

paulinejmaurice, 2020-12-11:
Hello,
One question about the "local frame = true" case.
You stated above that "The position error was expressed in the reference frame, and when converting it to the world frame I was instead assuming it was expressed in the actual local frame."
But in the local_frame case, m_p_error_vec (which is in the reference frame if I am correct) is used as is. Whereas the velocity and acceleration error terms are converted using "m_wMl.rotation(oMi.rotation())" (l. 187), so they are converted from the world to the actual local frame. Does that mean position error is in the reference frame and velocity/acc errors are in the actual local frame? Or am I missing something? Thanks.

paulinejmaurice, 2020-12-11:
About the difference between 
cout<<"m_p_error_vec="<<m_p_error_vec.head<3>().transpose()<<endl;
cout<<"oMi-m_M_ref  ="<<(oMi.translation()-m_M_ref.translation()).transpose()<<endl;

I copy-pasted that as you suggested, and I tried on my humanoid model (origin in (0,0,0)) with a task on the right hand. I do observe difference between the 2 results, even though the reference is not totally crazy (end reference is (0.5, 20, 0.8) which is close to the start position except for the Y coordinate). I have a mask in order to control only the position in Z (no control of the orientation).
Examples of the values I observe are:

m_p_error_vec=    0.128302     -2.46898    -0.000202907
oMi-m_M_ref  =  0.125921   -2.47089    0.00236986
R
 0.856283 -0.227082  0.460507
 0.124388   0.96216  0.240721
-0.498157 -0.147826    0.8525
p
-0.186367
  1.98838
 0.800009

andreadelprete, 2020-12-11:
> Hello,
> One question about the "local frame = true" case.
> You stated above that "The position error was expressed in the reference frame, and when converting it to the world frame I was instead [...]

You are right. I had noticed that as well the other day, but I could only fix it today for lack of time. It has been fixed 5 hours ago in my last commit on the devel branch.

andreadelprete, 2020-12-11:
> About the difference between
> cout<<"m_p_error_vec="<<m_p_error_vec.head<3>().transpose()<<endl;
> cout<<"oMi-m_M_ref ="<<(oMi.translation()-m_M_ref.translation()).transpose()<<endl;
>
> I copy-pasted [...]

These differences are too large. There seems to be a problem. First, are you using the latest devel branch? If not, could you make sure you still see these differences with the latest version of the code?
If so, could you provide me with a script to reproduce the problem? If you can reproduce it with one of the standard robots that come with TSID or example-robot-data that would ease everything.

paulinejmaurice, 2020-12-12:
I was indeed using the devel branch, but only with the first fix (the one from 2 days ago). With your commit from yesterday, now it's working all fine! Thanks!

dalinel, 2020-12-12:
I knew that @paulinejmaurice had the same kind of problem which was not a numerical issue
Now that everything is fine I also think this is good to leave it like this
Thanks a lot for the commits !

andreadelprete, 2020-12-12:
Great, I'm glad everything is working now. Thanks to you for pointing out this problem!

closed 2020-12-12 by andreadelprete