stack-of-tasks/tsid #121: pinoccho compatible issue

ddliu365, 2020-12-27:
Hi,

When I updated the `pinocchio` and found the tsid importation issue like the following:

`ImportError: libpinocchio.so.2.4.6: cannot open shared object file: No such file or directory`

Is it possible to solve this issue without compiling it?

andreadelprete, 2020-12-30:
As first thing I would try to reinstall tsid:
```
sudo apt install --reinstall robotpkg-tsid
```

nim65s, 2020-12-30:
your TSID is linked to pinochio v2.4.6, while your installed version is the up-to-date v2.5.4. I'd go for a `sudo apt update && sudo apt upgrade`.

ddliu365, 2020-12-30:
Thanks for your reply. It seems not caused by this issue.

For `crocoddyl` and `pinocchio`, they also have been updated but there is not issue.

I tried the above method, but the error still pops up.

```
jack@desktop:~$ python
Python 2.7.12 (default, Oct  5 2020, 13:56:01) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import pinocchio
>>> import crocoddyl
>>> import tsid
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/tsid/__init__.py", line 1, in <module>
    from .libtsid_pywrap import *
ImportError: libpinocchio.so.2.4.6: cannot open shared object file: No such file or directory

```

ddliu365, 2020-12-30:
When I try to clone the repo and it shows the error:
```
jack@desktop:~/repos$ git clone --recursive git@github.com:stack-of-tasks/tsid.git
Cloning into 'tsid'...
Warning: Permanently added the RSA host key for IP address '140.82.113.3' to the list of known hosts.
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
```

andreadelprete, 2020-12-30:
Could you show us the output of `dpkg -l | grep robotpkg` to check if the versions of pinocchio and tsid have been updated?

ddliu365, 2020-12-30:
This is the information.

```
jack@desktop:~$ dpkg -l | grep robotpkg
ii  robotpkg-collada-dom                            2.4.0r1                                               amd64        C++ object representation of a COLLADA XML instance document
ii  robotpkg-curves                                 0.5.2                                                 amd64        Library for creating curves
ii  robotpkg-dynamic-graph-v3                       4.3.1                                                 amd64        Efficient data-flow C++ library for robotics
ii  robotpkg-eigen-quadprog                         1.0.1r1                                               amd64        QuadProg QP solver using the Eigen3 library
ii  robotpkg-eiquadprog                             1.2.2                                                 amd64        algorithm of Goldfarb and Idnani for the solution of a QP
ii  robotpkg-example-robot-data                     3.7.0                                                 amd64        Robot URDFs for benchmarking and developed examples
ii  robotpkg-hpp-fcl                                1.6.0                                                 amd64        Humanoid Path Planner (collision detection and distance computations)
ii  robotpkg-libccd                                 2.0                                                   amd64        collision detection
ii  robotpkg-octomap                                1.9.0                                                 amd64        An Efficient Probabilistic 3D Mapping Framework Based on Octrees
ii  robotpkg-openscenegraph                         3.6.5r3                                               amd64        open source high performance 3D graphics toolkit
ii  robotpkg-parametric-curves                      1.4.0                                                 amd64        This packages provides a template based parametric curves library.
ii  robotpkg-pinocchio                              2.5.4                                                 amd64        Efficient rigid body dynamics
ii  robotpkg-py27-crocoddyl                         1.5.0r1                                               amd64        Optimal control library for robot under contact sequence
ii  robotpkg-py27-curves                            0.5.2                                                 amd64        Library for creating curves (python bindings)
ii  robotpkg-py27-eigenpy                           2.5.0                                                 amd64        Eigen bindings to python
ii  robotpkg-py27-example-robot-data                3.7.0                                                 amd64        Robot URDFs for benchmarking and developed examples
ii  robotpkg-py27-hpp-fcl                           1.6.0                                                 amd64        Humanoid Path Planner (collision detection and distance computations (python bindingss))
ii  robotpkg-py27-parametric-curves                 1.4.0                                                 amd64        This packages provides a template based parametric curves library (python bindings)
ii  robotpkg-py27-pinocchio                         2.5.4                                                 amd64        Efficient rigid body dynamics (python bindings)
ii  robotpkg-py27-qt4-gepetto-viewer                4.10.1r2                                              amd64        Humanoid Path Planner (Simple viewer based on OpenSceneGraph)
ii  robotpkg-py27-qt4-gepetto-viewer-corba          5.5.1                                                 amd64        Humanoid Path Planner (Corba server for gepetto-viewer)
ii  robotpkg-py27-quadprog                          0.1.6r1                                               amd64        Solve a strictly convex quadratic program
ii  robotpkg-qt4-osgqt                              3.5.7r2                                               amd64        open source high performance 3D graphics toolkit
ii  robotpkg-simple-humanoid-description            1.0.2                                                 amd64        Rigid body model of a simple humanoid robot.
ii  robotpkg-tsid                                   1.4.2                                                 amd64        Efficient Inverse Dynamics based on Pinocchio

```

ddliu365, 2020-12-30:
I think I figured out the problem.

I just built tsid from source and it worked well now.

nim65s, 2020-12-30:
```
  File "/usr/local/lib/python2.7/dist-packages/tsid/__init__.py", line 1, in <module>
    from .libtsid_pywrap import *
 ```
 
You were importing a TSID build from source and not the binary one. And that one was built on an older version of pinocchio which was upgraded afterwards. re-building it fixed your issue, but the next time pinocchio binaries will update, you'll get the same issue.
 
 So if you don't need to build TSID from source, you should just remove this its manual installation in `/usr/local`. If you need it… Well don't forget to re-build and re-install it when a dependency is updated :)

closed 2020-12-30 by nim65s