stack-of-tasks/tsid #122: Getting NaN in joint accelerations by chance

NahuelVilla, 2021-02-24:
Hi to everyone, I open this issue because I found a strange behavior using TSID:

I have a simulation that uses TSID to control the robot Talos. When I run it, there is a probability that the function getAccelerations(HQP_solution) works producing correct joint accelerations and there is a probability for it to give, instead, a vector of nan.

By running the simulation again, and again I obtain either the correct behavior expected or an error resulting from those nan joint accelerations.

The problem is solved when I do not include tasks on position nor orientation of the base_link. I guess it can be related to the masks that I am setting on those tasks as the function setMask(vector) receives vectors of whatever size without complaining. Anyway, I have set masks with different sizes, in all cases, maintaining the same stochastic behavior on the solver.

Finally I made a repository with a minimum example for this problem:

https://gitlab.laas.fr/nvilla/tsid_problems/-/blob/master/Tasks_problems/problem_with_tasks.py#L9

In which "problem_with_tasks.py" can be executed showing the behavior for one single simulation and makes the statistics for a lot of N simulations.
Could someone tell me what is wrong?

Thanks,
Nahuel

andreadelprete, 2021-02-25:
Hi @NahuelVilla , I've just tested your script, but on my system it works correctly. I had to change a few things, probably because I'm using a different version of TSID. For instance, in my version, the method to set the mask of a task is `setMask` and not simply `mask`. Since in December I had fixed a few bugs but these changes haven't been ported to master yet, that's probably the reason why you get errors and I don't. 

Maybe @nim65s could tell us when the next release of TSID is planned? Otherwise you could just switch to the latest devel version of TSID, which is anyway a good thing to do for you if you wanna benefit immediately from the latest features as soon as they are implemented.

nim65s, 2021-02-25:
Thanks @andreadelprete for the feedback, I'll make a release and check if this fix @NahuelVilla issue.

NahuelVilla, 2021-03-01:
Thanks @andreadelprete. So, I will update my version.

andreadelprete, 2021-03-08:
@NahuelVilla can you close the issue if the problem has been solved?

NahuelVilla, 2021-03-09:
Thanks, the last update solved this problem.

I am closing the issue now.

closed 2021-03-09 by NahuelVilla