stack-of-tasks/tsid #125: Center of Pressure Task

andreadelprete, 2021-03-09:
I'm currently working on a new center of pressure (CoP) task for TSID. Currently we have a center of mass task and an angular momentum task, but no CoP task. This has been requested by @NahuelVilla and it could probably be useful for anybody working with a Linear Inverted Pendulum model. I open this issue to keep track of the developments and advertise this new feature.

andreadelprete, 2021-03-10:
I've just finished the first successful test of the CoP task. Starting from the `ex_4_walking.py` script (from the exercise folder) I've added a CoP task to track the reference CoP (besides tracking the reference CoM). Before introducing the task, this was the CoP I was getting:
[image]

Now with the CoP task it seems that the real CoP is much closer to the reference one:
[image]

I cannot get the tracking to be better than this because if I increase too much the weight of the CoP task then the robot falls. This is reasonable because I am setting an open-loop reference CoP, which is not stabilizing.

olivier-stasse, 2021-03-10:
It sounds very nice. Thanks a lot @andreadelprete  !

andreadelprete, 2021-03-10:
The CoP task is implemented and pushed to devel now, so I'm closing this issue.

closed 2021-03-10 by andreadelprete