stack-of-tasks/tsid #131: Unilaterality constraint not working

NahuelVilla, 2021-03-22:
Hello, 

I obtained negative ground contact forces with TSID as can be seen in the following script.
https://gitlab.laas.fr/nvilla/tsid_problems/-/blob/master/NegativeForces/simu.py#L14

The problem appears when a foot in contact with the ground is not required to support forces. The script shows this problem in both cases, when setting directly the reference forces and when using the new CoP task.

On the CoP task, moreover, when it is set as a constraint (priority level 0), the robot falls directly to the ground. I guess that the vertical reference force has been kept at zero, which may be the reason of this failure.

Thanks for the help with these questions.

andreadelprete, 2021-03-22:
I' ve tried running your code but I get this error:
```
Traceback (most recent call last):
  File "/home/student/devel/src/tsid/exercizes/tsid_problems/NegativeForces/simu.py", line 9, in <module>
    from motion_generation import MotionReader
  File "/home/student/devel/src/tsid/exercizes/tsid_problems/NegativeForces/motion_generation.py", line 3, in <module>
    import curves
ImportError: No module named 'curves'
```
If I comment the import then I get this error:
```
Traceback (most recent call last):
  File "/home/student/devel/src/tsid/exercizes/tsid_problems/NegativeForces/simu.py", line 11, in <module>
    from wbc import WBController
  File "/home/student/devel/src/tsid/exercizes/tsid_problems/NegativeForces/wbc.py", line 10, in <module>
    from biped_tsid import BipedActuators
  File "/home/student/devel/src/tsid/exercizes/tsid_problems/NegativeForces/biped_tsid.py", line 18, in <module>
    from qpsolvers import osqp_solve_qp
ImportError: No module named 'qpsolvers'
```
If I comment also this import I get this error:
```
Traceback (most recent call last):
  File "/home/student/devel/src/tsid/exercizes/tsid_problems/NegativeForces/simu.py", line 30, in <module>
    motion_generator = MotionReader(conf)
  File "/home/student/devel/src/tsid/exercizes/tsid_problems/NegativeForces/motion_generation.py", line 34, in __init__
    self.concatenate_all()
  File "/home/student/devel/src/tsid/exercizes/tsid_problems/NegativeForces/motion_generation.py", line 63, in concatenate_all
    trajectories.update({'CoM': cs.concatenateCtrajectories()})
TypeError: No to_python (by-value) converter found for C++ type: ndcurves::piecewise_curve<double, double, true, Eigen::Matrix<double, -1, 1, 0, -1, 1>, Eigen::Matrix<double, -1, 1, 0, -1, 1>, ndcurves::curve_abc<double, double, true, Eigen::Matrix<double, -1, 1, 0, -1, 1>, Eigen::Matrix<double, -1, 1, 0, -1, 1> > >
>>> 
```

I have already installed the following curves-related packages:
```
dpkg -l | grep curves
ii  robotpkg-ndcurves                           1.0.0                                           amd64        Library for creating curves
ii  robotpkg-parametric-curves                  1.4.0                                           amd64        This packages provides a template based parametric curves library.
ii  robotpkg-py35-ndcurves                      1.0.0                                           amd64        Library for creating curves (python bindings)
ii  robotpkg-py35-parametric-curves             1.4.0                                           amd64        This packages provides a template based parametric curves library (python bindings)
```
Any idea what's missing on my system?

nim65s, 2021-03-22:
Curves was renamed into NDCurves in v1.0.0 (https://github.com/loco-3d/ndcurves/releases/tag/v1.0.0 / https://github.com/loco-3d/ndcurves/issues/2) and it that `motion_generation.py` as not been updated for this change.

@NahuelVilla : you'll have to update your scripts with this. Do you need help for it ?

@andreadelprete : meanwhile, you can just replace `curves` by `ndcurves`, as you already have the right packages installed.

PS: @andreadelprete : those `-py35-` on your system show that you are using Ubuntu 16.04, which will not be supported by Canonical after April 2021.

NahuelVilla, 2021-03-22:
Sorry I should have mentioned it at the beginning. 
Curves is used to read the motion saved in the contact sequence file. I am not using it directly and changing it by ndcurves should be fine. 

The force distributor is solving a QP problem using OSQP which can be installed from
https://pypi.org/project/osqp/

And this QP solver is managed using the library of Stephane Caron, qpsolvers, which can be installed from this link:
https://pypi.org/project/qpsolvers/

Also sympy is needed, but probably you already have it installed.

Thanks

andreadelprete, 2021-03-22:
I've added this line to see how the QP looks like:
```
HQPData.print_all()
```
and it looks like this:
```
Level 0
 - base-dynamics: w=1, equality, 6x62
 - task-right-foot: w=100, equality, 6x62
 - task-left-foot: w=100, equality, 6x62
 - task-orientation-root: w=10, equality, 3x62
 - contact_rfoot_motion_task: w=1, equality, 6x62
 - contact_rfoot_force_constraint: w=1, inequality, 17x62
 - contact_lfoot_motion_task: w=1, equality, 6x62
 - contact_lfoot_force_constraint: w=1, inequality, 17x62
Level 1
 - task_cop: w=1, equality, 3x62
 - task-posture: w=0.1, equality, 32x62
 - task-position-root: w=100, equality, 3x62
 - task-am: w=0.02, equality, 3x62
 - contact_rfoot_force_reg_task: w=0.1, equality, 6x62
 - contact_lfoot_force_reg_task: w=0.1, equality, 6x62
```
It seems to me that you have redundant constraints because  task-right-foot and contact_rfoot_motion_task are the same thing. Eiquadprog doesn't handle redundant equality constraints, so I guess that's the cause of the problem.

andreadelprete, 2021-03-22:
Indeed if you set `level_foot = 1` the problem disappears.

andreadelprete, 2021-03-22:
> Curves was renamed into NDCurves in v1.0.0 (https://github.com/loco-3d/ndcurves/releases/tag/v1.0.0 / [loco-3d/ndcurves#2](https://github.com/loco-3d/ndcurves/issues/2)) and it that `motion_generation.py` as not been updated for this change.
> 
> @NahuelVilla : you'll have to update your scripts with this. Do you need help for it ?
> 
> @andreadelprete : meanwhile, you can just replace `curves` by `ndcurves`, as you already have the right packages installed.
> 
> PS: @andreadelprete : those `-py35-` on your system show that you are using Ubuntu 16.04, which will not be supported by Canonical after April 2021.

Thanks for the heads up!

NahuelVilla, 2021-03-22:
Yes! This solved the problem. 

Thanks!

For some reason HQPData.print_all() don't print anything to me :/, but I guess it is not important.
Still, when using the copTask as a constraint it makes the robot disappear in this script now. Could it have a similar reason?

Thanks a lot for answering so fast.

nim65s, 2021-03-23:
I think we can close the issue then :)

closed 2021-03-23 by nim65s