stack-of-tasks/tsid #132: Collecting TSID applications

andreadelprete, 2021-04-29:
Hi everybody,

I've been asked to give a presentation on TSID (on Monday), during which I'd like to show how TSID has been used in these 4 years. If you've used TSID, either in simulation or on real hardware, could you provide me with a short description of:
* the robot (name, lab, morphology, number of DoFs)
* the task (e.g., bipedal walking, trotting, jumping, balancing)
* link to paper and/or video showing the resulting robot motion

This information is gonna be useful also for a future journal paper on TSID, which I plan to write eventually.
Thanks in advance!

NoelieRamuzat, 2021-04-29:
Hi Andrea !
I will be glad to provide you my results with TSID:
- We use the robot Pyrene at the LAAS-CNRS, a TALOS humanoid robot from PAL Robotics. It has 32 joints + 6 DoFs for the floating base.
- In simulation we achieved bipedal walking, climbing stairs and walking on tilted platforms (using the Angular Momentum task for better results). On the real robot for now only a simple posture task with feet contacts has been tested successfully. 
- Here is a link to a work in pogress paper with the results: 
[Comparison of Position and Torque Whole Body Control Schemes on the TALOS Humanoid Robot](https://hal.archives-ouvertes.fr/hal-03145141) 
and the associated video:
https://cloud.laas.fr/index.php/s/0q1ZgY55uMxOfjl

Best !

paLeziart, 2021-04-29:
Hello @andreadelprete ,

* Solo-12 at LAAS-CNRS, quadruped robot developed by several teams as part of the [Open Dynamic Robot Initiative](https://open-dynamic-robot-initiative.github.io/), 12 DoF (motors) + 6 if you consider the free flyer
* Trotting in simulation to compare the performance of several model predictive controllers
* Article: [Comparison of predictive controllers for locomotion and balance recovery of quadruped robots](https://hal.laas.fr/hal-03034022)
* Video: https://peertube.laas.fr/videos/watch/d6d90690-3262-4e61-9d03-08a2623726e3 

Best,

ABonnefoy, 2021-05-02:
Hello @andreadelprete , 
sorry for the late answer, but here are some motions generated using TSID:

- Solo-12 at LAAS-CNRS was used, a 12 DoF quadruped developed within the [Open Dynamic Robot Initiative](https://open-dynamic-robot-initiative.github.io/) as stated above
- Some motions were developed using TSID, among which a push up and a jumping push up motion. So far, these motions have only been tested in Simulation, but they are intended to be tested on the real robot using Pierre-Alexandre Leziart's impedance controller. The ultimate goal is to use these motions as steps leading to jumping with the real robot.
- Simulation videos: https://cloud.laas.fr/index.php/s/G2w5pI2BGSSBMFy

Best!

andreadelprete, 2021-05-03:
>     * Trotting in simulation to compare the performance of several model predictive controllers

@paLeziart In the video I see there are also some real hardware experiments. Was TSID used for the experiments as well? Or was it used only for the simulations?

pFernbach, 2021-05-03:
Hello, I guess I missed the deadline for your presentation but you could still use this for the paper:

We integrated TSID in the [multicontact-locomotion-planning framework](https://github.com/loco-3d/multicontact-locomotion-planning). Inside this framework TSID is used to generate motion from references like contact sequence, centroidal trajectories, feet trajectories, ect coming from various methods from the state of the art. We are focusing on walking motions in non coplanar environments, so far it have been used to generate motions for:

* Solo 8 and 12 (only on flat floor, tested on the real solo8 and in simulation only for solo12)
* Anymal (Simulation only, exemple of motion: https://cloud.laas.fr/index.php/s/LkdZGgDypM0uxjJ)
* HRP-2 (On the real robot, exemple of motion: https://cloud.laas.fr/index.php/s/BlC3Cv4iuMdzbQz)
* Talos (Simulation only, exemple of motion: https://cloud.laas.fr/index.php/s/pdio5HHW1qiPm3Z)

It was also used to generate the first dataset of motion for Talos and Anymal for the MEMMO project. Exemple of motions in this dataset for Talos on flat floor :https://cloud.laas.fr/index.php/s/2ywi91okwTVYJdE

paLeziart, 2021-05-03:
> > ```
> > * Trotting in simulation to compare the performance of several model predictive controllers
> > ```
> 
> @paLeziart In the video I see there are also some real hardware experiments. Was TSID used for the experiments as well? Or was it used only for the simulations?

@andreadelprete For the hardware experiment at the end of the video the control of the robot was a replay. The desired torques and desired positions/velocities of actuators for each time step were saved during a simulation using TSID and PyBullet. Then, the log file was read and the torques sent to the robot were `tau = tau_des + P * (q_des - q) + D (v_des - v)`.

melya30, 2021-05-03:
Hello Andrea,
      
- Skeletal model with 42 DoFs (model: https://hal.archives-ouvertes.fr/hal-01841355)
-  Simulation of human-like gait for 9 individuals and 9 different gait models.
-  Article: Human-like gait generation from a reduced set of tasks using the hierarchical control framework from robotics:
https://hal.archives-ouvertes.fr/hal-02455162v1

andreadelprete, 2021-05-11:
Thanks everybody for the answers. In case you wanna see the presentation, here it is: https://www.youtube.com/watch?v=psy4uhi76Ko
I'm gonna close this issue now, but I'll come back to it when starting to prepare the paper on TSID. Keep you posted!

closed 2021-05-11 by andreadelprete