stack-of-tasks/tsid #133: conda installation for python bindings

luyangzhu, 2021-05-14:
Hi Andrea, thanks for this wonderful project. I am asking if it is possible to add support for conda installation (like pinocchio, eigenpy and Crocoddyl) ?

andreadelprete, 2021-05-14:
Honestly I have no idea how much work that would require. Maybe @nim65s knows?

nim65s, 2021-05-14:
As a matter of fact, @jcarpent and his team are responsible for the conda packaging of projects in the SoT, HPP, gepetto & loco-3d namespaces, so maybe he should answer here.

I forgot to mention it here, but [TSID is available there](https://pypi.org/project/tsid/). So @luyangzhu, if you are on Linux, you can use the wheels I'm publishing on PyPI. This is a first version with limited features, but from a user point of view, everything should work with a `pip install tsid`.

I'll add wheels for OSX soon. And for Windows, well, I didn't even start to think about it yet.

luyangzhu, 2021-05-14:
Thanks. I will try the Pypi.

closed 2021-05-14 by luyangzhu