stack-of-tasks/tsid #134: Issues with importing tsid in python and conda environment

Neotriple, 2021-05-24:
Hi there,

I'm currently running a few of the LAAS projects (pinocchio, crocoddyl, and TSID) for some of my work. In my current setup, I've installed Crocoddyl and Pinocchio (and their accompanying software) through conda, and TSID from source. The issue I'm currently running into is that I can't import both crocoddyl *and* tsid at the same time. i.e. I can't do an:

```
import crocoddyl
import tsid
```

at the top of my python file. Whichever import comes second will give me an associated error from the pywrap, i.e. for the above, I get the following:

`File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.9/site-packages/tsid/__init__.py", line 1, in <module>
    from .libtsid_pywrap import *`

(And for reference, I do know that tsid isn't in the appropriate conda site-packages folder, so I added the above site-packages to my path manually since it did not install to the correct place).

If I instead do:

```
import tsid
import crocoddyl
```

I'll get

`File "<stdin>", line 1, in <module>
  File "/home/pshah/anaconda3/envs/biconvex_mpc/lib/python3.9/site-packages/crocoddyl/__init__.py", line 1, in <module>
    from .libcrocoddyl_pywrap import *
SystemError: initialization of libcrocoddyl_pywrap raised unreported exception`

Any reason this would come up?

jcarpent, 2021-05-24:
tsid will be integrated soon within conda. We have recently introduced EigenQuadProg, the missing component.
What you need to do is to install tsid within your conda environment. This is very easy as:
```
cmake .. -DCMAKE_INSTALL_PREFIX=$CONDA_PREFIX -DBUILD_TYPE=Release
```
Your issue is that you cannot mix various Python versions. Here, you mix conda and classic Python site-packages, which should be avoided. This is the purpose of conda in fact.

Neotriple, 2021-05-29:
Hi, just wanted to reopen this issue; I am still having the same issue despite the change made above. In order to get a clean install I removed the build folder and created it again.

jcarpent, 2021-05-30:
Sorry @Neotriple, but your message is not explicit. From what we can guess, you don’t have a clean install. Please see with your colleagues at Oxford to get local help.

closed 2021-05-30 by jcarpent