stack-of-tasks/tsid #137: Tsid for position control in simulation

ddliu365, 2021-06-13:
Hi, I try to use tsid for a position controlled robot in Pybullet simulation.

It doesn't work properly while the torque controlled robot works well.

I am not sure if I made a mistake here or not.

Here is the code in my control loop.
```
tsid.comTask.setReference(tsid.trajCom.computeNext())
tsid.postureTask.setReference(tsid.trajPosture.computeNext())
tsid.rightFootTask.setReference(tsid.trajRF.computeNext())
tsid.leftFootTask.setReference(tsid.trajLF.computeNext())
HQPData = tsid.formulation.computeProblemData(t, q.copy(), dq.copy())
sol = tsid.solver.solve(HQPData)

if useTorqueCtrl:
	tau = tsid.formulation.getActuatorForces(sol)
else:
	ddq = tsid.formulation.getAccelerations(sol)
        q, dq = tsid.integrate_dv(q, dq, ddq, conf.dt)
	tau = q[7:]
tsid.display(q)

bipd.send_joint_command(tau) # if useTorqueCtrl=True, then simulator will accept torque commands, 
                                                  #  otherwise, it will accept position commands

p.stepSimulation()
q, dq = bipd.get_state()
t +=dt
```

I test the stabilizer by apply a 500N force from the side of torso and it is applied during 2ms. For torque controlled robot, tsid can stabilize the robot well. For a position controlled robot, the foot will swing left and right and there is no observed stabilization result in the simulation.

I also increased the weights for the foot task, however, the result is similar to the one without any controller. Should I add a motion task for the torso? It seems the swing of torso is most remarkable during the position controlled robot when it is applied by disturbance

How to generate a compliant behavior like torque controlled robot on a position controlled robot by using tsid? Maybe my question is stupid here, but can you help to address the problem or give some hint for this application scenario? Thanks.

andreadelprete, 2021-06-16:
I am not sure, but maybe the problem is this line:
```
q, dq = bipd.get_state()
```
By "reading" the state of the robot from the simulator and using it then in TSID, you are computing the desired accelerations based on the real state, but then you're integrating these accelerations over the "integrated state" (i.e., the one that you get by calling `integrate_dv`). You can try commenting that line to see if things improve.

ddliu365, 2021-06-16:
sure, I will do it. But if I remove the state from simulator, how tsid `feels` the disturbance from the environment in the simulator?

andreadelprete, 2021-06-16:
It doesn't. That's the problem with position control.

ddliu365, 2021-06-30:
Yes, it seems a big issue for position controlled robot.

Right now I have an idea. First I simulated a torque controlled robot by using tsid(same physical properties, limit etc as position controlled robot) and then roll out the joint position trajectory. Until now, everything is offline.

Then I would like to use an online "task space inverse kinematics controller" to command the physical robot to track the trajectory by adding kinematics tasks, such as torso orientation, posture task..., based on state estimation from sensors(imu, foot contact, encoders)

The "task space inverse kinematics controller", I mean solving inverse kinematics in a quadratic programming similar to the idea you mentioned in TSID.

Do you think TSID can be extended like this way?

andreadelprete, 2021-07-01:
If you mean an acceleration-level inverse kinematics, then yes, TSID can be used for that. However, I see no reason for using kinematics instead of dynamics. By using dynamics you can ensure that contact forces are within friction cones and joint torques are within torque limits. With kinematics you cannot.

ddliu365, 2021-07-01:
Yes. I actually tried to use TSID directly on a position controlled robot and it doesn't work well. Maybe I messed up something. That's why I want to use TSID indirectly first, that is, to use TSID as an offline controller to obtain a dynamically feasible trajectory. However, this trajectory cannot be directly implemented on a position controlled robot because of motor backlash, or part compliance, etc. Thus, first order or second order inverse kinematics will be used as an online controller to command robot to follow the trajectory filtered by offline TSID. Do you think it is feasible?

andreadelprete, 2021-07-30:
Yes, I think it could work. Not sure how you wanna use TSID though.

andreadelprete, 2022-01-26:
Since nobody has written in this issue for some months now, I think we can close it now. Feel free to re-open if needed.

closed 2022-01-26 by andreadelprete