stack-of-tasks/tsid #142: signature C++ problem with boost.python

NiaoBochen, 2021-08-25:
With robotpkg install and robotpkg-py35-tsid when trying to do
`python3 tests/python/test_Solvers.py`
I have the following error:
`Test Solvers

Gonna perform 100 tests with 60 variables,  36 equalities 40 inequalities
Traceback (most recent call last):
  File "tests/python/test_Solvers.py", line 59, in <module>
    const1.append(1.0, eq_const)
Boost.Python.ArgumentError: Python argument types in
    ConstraintLevel.append(ConstraintLevel, float, ConstraintEquality)
did not match C++ signature:
    append(tsid::python::ConstraintLevels {lvalue}, double, std::shared_ptr<tsid::math::ConstraintBound> data)
    append(tsid::python::ConstraintLevels {lvalue}, double, std::shared_ptr<tsid::math::ConstraintInequality> data)
    append(tsid::python::ConstraintLevels {lvalue}, double, std::shared_ptr<tsid::math::ConstraintEquality> data)
`
My environment : 
Ubuntu 16.04.7 LTS, 
python3.5
robotpkg-py35-pinocchio/xenial,now 2.6.3
robotpkg-py35-tsid/xenial,now 1.6.0

How can i fix it?

Thanks!

nim65s, 2021-08-25:
Hi,

The main issue is that you are using an obsolete distribution. Please upgrade to 18.04 or 20.04 which are still supported.

But anyway, on 18.04, we have another issue with this test:

```python
python3 tests/python/test_Solvers.py

Test Solvers

Gonna perform 100 tests with 60 variables,  36 equalities 40 inequalities
check constraint level #0
 - eq1: w=1, equality, 36x60
 - ini1: w=1, inequality, 40x60

check constraint level #1
 - c1: w=1, equality, 60x60

Check HQP DATA
Level 0
 - eq1: w=1, equality, 36x60
 - ini1: w=1, inequality, 40x60
Level 1
 - c1: w=1, equality, 60x60

Segmentation fault (core dumped)
```

I deactivated this test in https://github.com/stack-of-tasks/tsid/commit/0c876f207097cfdd9bade940ebde8d22b9f042f5 as it is not working.

I'll double-check what's wrong with it.

NiaoBochen, 2021-08-25:
Thanks for your reply, I will try the updated version of Ubuntu

andreadelprete, 2022-01-26:
Since we haven't heard back for a few months I think we can close this issue now.

closed 2022-01-26 by andreadelprete