stack-of-tasks/tsid #145: Adding IK to tsid

EtienneAr, 2021-10-04:
Hello, 

We would like to do proper IK control on our robots at INRIA Paris (with @jcarpent @proyan).
We were thinking that the best way to do it, would be to add IK features to tsid as most of the code could be re-used.

But we were wondering what is the best way to add such feature in tsid ? (We will have to re-implement some tasks classes etc... But those classes are already named with a generic name, and not a specific names for "dynamic").

Thus one approach could be to simply add the suffix IK to the name of the reimplemented classes (like TaskMotionIK).
Or another approach could be to create a new namespace tsik and define the new classes inside.

What is your opinion @andreadelprete ?

andreadelprete, 2021-10-04:
I could be wrong, but I think you don't need to reimplement the task classes if you wanna do acceleration-based IK. You just need to use a different implementation of InvDynFormulationUtils, which doesn't account for contact forces. 

If instead you wanna do velocity-based IK, then yes, you'd need to reimplement the task classes. In this case adding the suffix IK to the class name would be fine to me.

EtienneAr, 2021-10-05:
Ok.
I want to do velocity-based IK, so I will add the suffix to tasks class name.
Thank you.

closed 2021-10-05 by andreadelprete