stack-of-tasks/tsid #146: Cannot run the demo files

BolunDai0216, 2021-10-08:
Hi,

When I was running `demo_quadruped.py` and `demo_romeo.py`, I encountered the following error:
```
############################################################
############## Test TSID with Quadruped Robot ##############
############################################################ 

libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
/opt/openrobots/lib/python3.8/site-packages/pinocchio/robot_wrapper.py:272: UserWarning: Error while starting the viewer client.
Check whether gepetto-viewer is properly started
  self.viz.initViewer(*args, **kwargs)
Traceback (most recent call last):
  File "demo/demo_quadruped.py", line 82, in <module>
    robot_display.displayCollisions(False)
  File "/opt/openrobots/lib/python3.8/site-packages/pinocchio/robot_wrapper.py", line 307, in displayCollisions
    self.viz.displayCollisions(visibility)
  File "/opt/openrobots/lib/python3.8/site-packages/pinocchio/visualize/gepetto_visualizer.py", line 204, in displayCollisions
    nodeName = self.getViewerNodeName(collision,pin.GeometryType.COLLISION)
  File "/opt/openrobots/lib/python3.8/site-packages/pinocchio/visualize/gepetto_visualizer.py", line 23, in getViewerNodeName
    return self.viewerCollisionGroupName + '/' + geometry_object.name
AttributeError: 'GepettoVisualizer' object has no attribute 'viewerCollisionGroupName'
```
Does anyone know what might be the issue?

Additionally, I could not import `from pinocchio import libpinocchio_pywrap as pin`, since it is not actually used, I just commented it out. Should I be able to import it?

Thanks in advance.

nim65s, 2021-10-08:
Hi,

It looks like we have issues with those demos. I made some trivial fixes in #147, but that's not enough to get everything work, especially on the romeo side.

But from your error message, I would guess that  you have gepetto-viewer, but not gepetto-viewer-corba.
Could you run `sudo apt install robotpkg-py38-qt5-gepetto-viewer-corba` and try the quadruped demo again ?

BolunDai0216, 2021-10-08:
> Hi,
>
> It looks like we have issues with those demos. I made some trivial fixes in #147, but that's not enough to get everything work, especially on the romeo side.
>
> But from your error message, I [...]

I just checked and it shows that I do have gepetto-viewer-corba installed.
```
sudo apt install robotpkg-py38-qt5-gepetto-viewer-corba
Reading package lists... Done
Building dependency tree       
Reading state information... Done
robotpkg-py38-qt5-gepetto-viewer-corba is already the newest version (5.6.0).
0 upgraded, 0 newly installed, 0 to remove and 463 not upgraded.
```

BolunDai0216, 2021-10-12:
Is there a way that I can check whether if the geppeto viewer is working or not? I can see it flash and then disappear when ever I am trying to run the demo code.

nim65s, 2021-10-12:
Ok, the segfault explains why the gui launch, but then the viewer initialization fail. But I can't reproduce it with the binaries. So I guess there is something wrong with those binary packages on your system.

Could you `apt update` and `apt upgrade` your system and try again ? If the problem persist, can you show me the outcome of `dpkg -l | grep 'openscenegraph\|omni'` ?

BolunDai0216, 2021-10-13:
> I updated the system, however I left my office, is there a way to run `gepetto-gui` headless?
> 
> When updating the system all I see are ROS related updates, Chrome and Firefox, I assume it would not solve my problem? I will try again tomorrow.
> 
> Here is the outcome of `dpkg -l | grep 'openscenegraph\|omni'`
> 
> ```
> ii  robotpkg-omniorb                                4.2.4                                                                 
> amd64        ORB that implements the 2.6 specification of the CORBA
> ii  robotpkg-openscenegraph                         3.6.5r3                                                                
> amd64        open source high performance 3D graphics toolkit
> ii  robotpkg-py38-omniorbpy                         4.2.4                                                                  
> amd64        Robust high-performance CORBA ORB for Python
> ```

I just tried it, the same message is displayed
```
/opt/openrobots/lib/python3.8/site-packages/pinocchio/robot_wrapper.py:272: UserWarning: Error while starting the viewer client.
Check whether gepetto-viewer is properly started
  self.viz.initViewer(*args, **kwargs)
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/opt/openrobots/lib/python3.8/site-packages/example_robot_data/__main__.py", line 13, in <module>
    load(args.robot, display=not args.no_display)
  File "/opt/openrobots/lib/python3.8/site-packages/example_robot_data/robots_loader.py", line 584, in load
    return loader(name, display, rootNodeName).robot
  File "/opt/openrobots/lib/python3.8/site-packages/example_robot_data/robots_loader.py", line 578, in loader
    inst.robot.display(inst.robot.q0)
  File "/opt/openrobots/lib/python3.8/site-packages/pinocchio/robot_wrapper.py", line 303, in display
    self.viz.display(q)
  File "/opt/openrobots/lib/python3.8/site-packages/pinocchio/visualize/gepetto_visualizer.py", line 176, in display
    if self.display_collisions:
AttributeError: 'GepettoVisualizer' object has no attribute 'display_collisions'
```

andreadelprete, 2021-11-16:
Dear @BolunDai0216 , sorry for the delay. I've checked the demo, and actually it is broken. However, this demo is also redundant with respect to the scripts in the "exercises" folder, so I think the best thing to do is simply to remove it, to avoid future issues. Thanks for pointing this out to us.

closed 2021-11-16 by andreadelprete