stack-of-tasks/tsid #152: Choose floating-base (or not) when creating a robot wrapper

EtienneAr, 2021-11-23:
I wanted to create a robot from a pinocchio model using the constructor `RobotWrapper(const Model & m, bool verbose=false)` (or more precisely its Python binding).

However, the robot is considered to have a floating base by default and mine hasn't. I didn't find a way to change it using the current code, so I wrote those 2 lines of code : 7d795cf4a7293f25e0df4bd523b203082c104b51 .

Should I PR this change, or was there already a way to do it ?
If I PR, is this the good way to give the user a choice ?
Should I also update for the 2 other constructors (and bindings)?

andreadelprete, 2021-11-23:
Dear @EtienneAr , thanks for your message. There's already a way to use fixed-base robots with TSID:

https://github.com/stack-of-tasks/tsid/blob/master/exercizes/ex_0_ur5_joint_space_control.py#L25

You just need to set the third (optional) input argument of the constructor of RobotWrapper to false, as in the example above.

EtienneAr, 2021-11-23:
Yes, but this works only if you create a robot from a file.

In my case, I would like to create a robot wrapper from a pinocchio model (it doesn't rely on any urdf file).

There is only one constructor in this case, and it assumes a floating base : https://github.com/stack-of-tasks/tsid/blob/b0d6bff80292fb3451ca7ca438a4ab84b5b8c022/src/robots/robot-wrapper.cpp#L47

andreadelprete, 2021-11-23:
Ok, now I see. In this case you can do a PR to merge your contribution in TSID. However, I would suggest following the same approach used for the other constructor, that is to take as input parameter a `const pinocchio::JointModelVariant & rootJoint` (rather than a `bool`).

EtienneAr, 2022-08-30:
Closing since #154 has been merged.
Thank you.

closed 2022-08-30 by EtienneAr