stack-of-tasks/tsid #156: import error while "import tsid"

YunxiTang, 2022-01-26:
Hi,
I installed tsid through conda and I wan to have test to check whether it is well installed. 
In the virtual environment, I cannot import tsid properly and the error is as following:
'''
Python 3.7.11 (default, Jul 27 2021, 14:32:16) 
[GCC 7.5.0] :: Anaconda, Inc. on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import tsid
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/tang/anaconda3/envs/tsid_env/lib/python3.7/site-packages/tsid/__init__.py", line 1, in <module>
    from .libtsid_pywrap import *
ImportError: libpinocchio.so.2.6.3: cannot open shared object file: No such file or directory
'''
Any help is appreciated. Thanks!

andreadelprete, 2022-01-26:
Hi @YunxiTang , it seems you're not able to find the pinocchio library. I am not familiar with the conda installation, so probably @nim65s is better suited to answer this question.

nim65s, 2022-01-26:
Sorry, but I'm not familiar at all with it :/
But it looks like you are looking for pinocchio 2.6.3, while the latest is 2.6.4, so you probably need to update something

YunxiTang, 2022-01-27:
@andreadelprete @nim65s Thanks for your quick replies. I think I solved this issue by installing pinocchio 2.6.3 firstly and then re-installing tsid. Thanks again and I think I can close this issue.

closed 2022-01-27 by YunxiTang