stack-of-tasks/tsid #163: Alignment error with -march=native, fixed with c++17

jbmouret, 2022-03-30:
Hi

I am trying to get more performance by compiling with `-march=native` to get Eigen vectorization.

I am compiling with clang for now:
` CC=clang CXX=clang++ CXXFLAGS="-march=native" cmake .. -DCMAKE_PREFIX_PATH=/home/pal/install -DCMAKE_BUILD_TYPE=Debug -DCMAKE_INSTALL_PREFIX=/home/pal/install -DBUILD_PYTHON_INTERFACE=OFF`

=> the test tsid-formulation fails

```

~/tsid/build > tests/tsid-formulation  
Running 4 test cases...

*** test_invdyn_formulation_acc_force_remove_contact ***
q: 0 0 0 0 0 0 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
tsid-formulation: /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:128: Eigen::internal::plain_array<double, 36, 0, 32>::plain_array() [T = double, Size = 36, MatrixOrArrayOptions = 0, Alignment = 32]: Assertion `(internal::UIntPtr(array) & (31)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.
unknown location(0): fatal error: in "tsid_formulationTest/test_invdyn_formulation_acc_force_remove_contact": signal: SIGABRT (application abort requested)
/home/pal/tsid/tests/tsid-formulation.cpp(128): last checkpoint

*** test_invdyn_formulation_acc_force ***
q: 0 0 0 0 0 0 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
tsid-formulation: /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:128: Eigen::internal::plain_array<double, 36, 0, 32>::plain_array() [T = double, Size = 36, MatrixOrArrayOptions = 0, Alignment = 32]: Assertion `(internal::UIntPtr(array) & (31)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.
unknown location(0): fatal error: in "tsid_formulationTest/test_invdyn_formulation_acc_force": signal: SIGABRT (application abort requested)
/home/pal/tsid/tests/tsid-formulation.cpp(128): last checkpoint

*** test_contact_point_invdyn_formulation_acc_force ***
tsid-formulation: /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:128: Eigen::internal::plain_array<double, 36, 0, 32>::plain_array() [T = double, Size = 36, MatrixOrArrayOptions = 0, Alignment = 32]: Assertion `(internal::UIntPtr(array) & (31)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.
unknown location(0): fatal error: in "tsid_formulationTest/test_contact_point_invdyn_formulation_acc_force": signal: SIGABRT (application abort requested)
/home/pal/tsid/tests/tsid-formulation.cpp(517): last checkpoint
```

If we follow the link (http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html), one solution is to switch to c++-17, which seems to work. In CMakeList.txt: 
`CHECK_MINIMAL_CXX_STANDARD(17 ENFORCE)`

```
Level 1
 - contact_rfoot_force_reg_task: w=1e-05, equality, 6x61
 - contact_lfoot_force_reg_task: w=1e-05, equality, 6x61
 - task-com: w=1, equality, 3x61
 - task-posture: w=0.01, equality, 31x61

Time 0
  contact_rfoot err: 0  contact_lfoot err: 0    task-com err: 0.1       task-posture err: 0     v=0      dv=0
dv(37x1): [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]

### TEST FINISHED ###
v(37x1): [ 0.0001,    0.03,  -6e-06,   1e-09,  0.0002,   0.004,  -0.004,   -0.05, -0.0006, -0.0001,  0.0005,    0.05,  -0.004,   -0.05,  0.0005, -0.0001, -0.0006,    0.05,   0.007, -0.0009,   0.003,   1e-05,  0.0009,   4e-06,  0.0001,  -1e-05,  0.0002,   1e-05,  -8e-06,  -3e-05,  0.0006,   0.003,   8e-06,  0.0009,   4e-06,  0.0001,   4e-06]
Final   CoM position:   0.0219545 0.000104108    0.665915
Desired CoM position: 0.0219541       0.1  0.665915
[...]
```

Otherwise, maybe there is a `EIGEN_MAKE_ALIGNED_OPERATOR_NEW` that is missing, but I did not find it (yet).

=> should TSID switch to c++-17?

nim65s, 2022-03-30:
From my point of view, the oldest compiler we currently support at LAAS is GCC 7.5 (default compiler of ubuntu 18.04) which is fully compliant with C++17.

Therefore, I wouldn't mind upgrading that `CHECK_MINIMAL_CXX_STANDARD()`. 

If anybody objects, we'll have to spend some time on that webpage. This is not an issue if anybody claim anything, but if nobody care I'd prefer to raise the standard.

Meanwhile, @jbmouret, while you're at it, could you make a quick check with `-DCMAKE_CXX_STANDARD=17` and see if the unit test is happy ?

jbmouret, 2022-03-31:
Yes, compiling with -DCMAKE_CXX_STANDARD=17 and clang works (for both my test and my code).

We are also using Ubuntu 18.04 (official distribution for Talos), although our Tiago is actually in 16.04 (but we will need to update it), which might cause some issues...

I will try to check with the g++ of ubuntu 18.04 later today.

commit a40793b9 (2022-09-09): upgrade to C++17, fix #163
commit eab9a866 (2022-09-09): Merge pull request #172 from nim65s/devel

upgrade to C++17, fix #163