stack-of-tasks/tsid #167: COP_Task - Looking for the 'romeo_walking_traj_tsid.npz' file.

Sabotm, 2022-07-26:
Hi !
I try to use the COP task based on the example in [this file](https://github.com/stack-of-tasks/tsid/blob/master/exercizes/test_cop_task.py). I saw that for the _COP task_, **' DATA_FILE_TSID'**  is needed but I can't find the root file corresponding to _**'romeo_walking_traj_tsid.npz'**_ . I need it to understand what is clearly _x_RF_ref , x_LF_ref_ and _cop_ref._
Maybe it's my fault but if you can help with it, it would be great! :)
Thank you !

andreadelprete, 2022-07-26:
Hi @Sabotm . To generate the date file you could use the other scripts in the same folder: 
- [ex_4_plan_LIPM_romeo.py](https://github.com/stack-of-tasks/tsid/blob/master/exercizes/ex_4_plan_LIPM_romeo.py)
- [ex_4_LIPM_to_TSID.py](https://github.com/stack-of-tasks/tsid/blob/master/exercizes/ex_4_LIPM_to_TSID.py)

The first script uses [this Python library](https://github.com/ahmadgazar/LMPC_walking) to generate a reference CoM trajectory for the robot. The second script interpolates this trajectory (which comes at a much lower frequency than what is needed for typical TSID use) and generates reference foot trajectories. I hope this helps.

Sabotm, 2022-07-26:
Hi !
Thank you for your answer! I'll look at it!
The link to the Python library does not seem to work.

fabinsch, 2022-09-13:
Hey @Sabotm, I achieved to use the scripts by cloning this library here: https://github.com/machines-in-motion/lmpc_walking into the exercizes folder as LMPC_walking.

andreadelprete, 2022-10-26:
This issue is inactive since a while, so I close it, but feel free to reopen it if you need.