stack-of-tasks/tsid #173: contact-6D: unclear - could be pulling force ?

fabinsch, 2022-09-14:
Hello, thanks for your nice work in TSID. I am working on the examples and I have a doubt regarding the 6D contact.

Compared with the `contact-point`, I understand that you are linearizing the friction cone which is leading to four conditions ([l.78 - l.82](https://github.com/stack-of-tasks/tsid/blob/master/src/contacts/contact-point.cpp#L79)) to check plus additionally, you need to check whether the contact force is pointing in the right direction ([l.84](https://github.com/stack-of-tasks/tsid/blob/master/src/contacts/contact-point.cpp#L84)). This is why matrix B has 5 lines in this case.

For the `contact-6D`, you are adding 4 contact points, that's why the matrix B contains 4 blocks (4x3) of the linearized friction cone ([l.105](https://github.com/stack-of-tasks/tsid/blob/master/src/contacts/contact-6d.cpp#L105)). What I don't understand is the last line of the matrix B, which is just a horizontal stack of the 4 contact normals ([l.108 - l.111](https://github.com/stack-of-tasks/tsid/blob/master/src/contacts/contact-6d.cpp#L108)). 
In my eyes, this can lead to a problem: we are comparing if the sum of the contact normals is pointing in the right direction, but not every single one of them, right? I would have expected to have 4 additional lines, one for each of the contact points, instead of 1 line that should account for all of them. This would lead to a matrix B with 4x5=20 lines instead of 4x4+1=17 lines.

I would be very happy if you can help me to understand your choice or the point that I'm missing currently. Thanks a lot.

jcarpent, 2022-09-19:
> And the resulting 3D force [is constrained to be positive](https://github.com/stack-of-tasks/tsid/blob/ce649ac45fea63f86e21fad5dd295cb21154da53/src/contacts/contact-6d.cpp#L113) so no pulling force is allowed.

In fact, pulling seems to be allowed because having $f_1 + f_2 + f_3 + f_4 \geq 0$, does not enforce the fact that $f_i \geq 0$

andreadelprete, 2022-09-19:
The reason you have only 1 constraint on the normal force (rather than 4) is because: i) the linearized friction cone constraints are already preventing the forces on each vertex to be pulling, and ii) TSID allows you to specify a minimum net normal force for each 6D contact (this is a design choice, but I find it very reasonable). So in the end you'll have all forces that are pushing, and the net normal force that is above the user-defined threshold `fMin`.

jcarpent, 2022-09-19:
Ok. Sounds good to me.

closed 2022-09-19 by jcarpent

fabinsch, 2022-09-19:
Alright, thanks a lot for your explanations @andreadelprete and @egordv .