stack-of-tasks/tsid #185: Cannot surpress SolverHQuadProg output

sjauhri, 2022-11-30:
**Installation: tsid from conda-forge 1.6.0**

I am using the solver: 
tsid.SolverHQuadProgFast("qp solver")

When calling a 'solve' using this solver, a huge amount of information is printed on the terminal.

Is there any way to disable this?

jcarpent, 2022-11-30:
Thanks @sjauhri for raising this issue. Could you try with the latest release 1.6.3?
Could you provide a tiny example to try on my side?

sjauhri, 2022-11-30:
Hi @jcarpent 
Thanks for the quick response!
Yes, with version 1.6.3 the issue is resolved. Closing this issue now.

Just FYI, I was running: https://github.com/stack-of-tasks/tsid/blob/master/exercizes/ex_1_ur5.py

closed 2022-11-30 by sjauhri

jcarpent, 2022-11-30:
Super. Nice to get your feedback.