stack-of-tasks/tsid #199: conda install broken ?

MedericFourmy, 2023-03-01:
I get an error while importing `tsid` when installing the library with conda. For instance, in a python=3.8 the error message is:

```
>>> import tsid
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/mfourmy/miniconda3/envs/py38/lib/python3.8/site-packages/tsid/__init__.py", line 1, in <module>
    from .libtsid_pywrap import *
ImportError: libpinocchio.so.2.6.12: cannot open shared object file: No such file or directory
```

To reproduce:
```
conda create -n py38 python=3.8
conda activate py38
conda install tsid -c conda-forge
python3 -c "import tsid"
```

In this case, when installing tsid, conda tells me that the pinocchio package to be installed is:  
`  pinocchio          conda-forge/linux-64::pinocchio-2.6.17-py38hac5dcfb_0 `  
which is not the version tsid seems to be looking for.  
For python 3.8, if I downgrade the pinocchio version:  `conda install pinocchio=2.6.12`, I don't get the error anymore.

I have a checked it for python [3.7-3.11] and on several computers (Ubuntu 20.04).

fabinsch, 2023-03-01:
Hi, @nim65s it seems were are not taking the newest `pinocchio` into account in the tsid_feedstock. You think this could be solved by just re-rendering it ?

fabinsch, 2023-03-01:
Thanks to @traversaro s comment on the pinocchio-feedstock I already understand the problem that we have better. We should discuss how we handle it [here](https://github.com/conda-forge/pinocchio-feedstock/issues/94).

jcarpent, 2023-03-12:
The fix is in the pipe https://github.com/conda-forge/tsid-feedstock/pull/23

closed 2023-03-12 by jcarpent